qmk/qmk_firmware #6765: Unable to reset

slipperydippery, 2019-09-18:
Hi, 

I have a planck Rev 4 keyboard (atmega 32u4 chip) . Last week when I reset the board, I couldn't flash it anymore due to an error: **Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.**

I was told I should follow the ISP flashing guide. So I bought a Pro Micro, and have flashed that using the QMK Toolbox. After some trial and error I now have the board flashed, and have even been able to flash it with my custom keymap. (with much help from HardAsMagnets/Germ)

However, when I try to flash the board via the usb port (`make planck:silvernitrate:dfu`), I am unable to, because the reset button (and reset key combination) now turn the board off and on again, instead of resetting it and allowing it to be flashed.  

_Additional information:_
Because the QMK Toolbox GUI couldn't locate the correct port, I flashed the Pro Micro via command line with the following commands: 

`avrdude.exe -p atmega32u4 -c avrisp -U flash:w:"C:\Users\zeronothingzero\Keyboards\Toolbox\pro_micro_ISP_B6_10.hex":i -P COM8`

and later the planck: 

`avrdude.exe -p atmega32u4 -c avrisp -U flash:w:"C:\Users\zeronothingzero\Downloads\planck_rev4_silvernitrate.hex":i -P COM8`

yanfali, 2019-09-18:
Hi @slipperydippery you need to use a slightly different flag when generating an ISP image. `:production` otherwise it won't have a bootloader. What's going on is you ISP flashed a hex without a bootloader, so when you reset it can't find a bootloader and jumps straight into the firmware.

Try making a hex like: `make planck/rev4:default:default` and flash that on using the ISP. Then you should be able to use toolbox again.

slipperydippery, 2019-09-18:
@yanfali Thanks for the quick reply! Your answer makes sense.

I tried running that command, but I get the error: `No rule to make target 'production'`. 

I also tried `make planck:default:production` and `make planck/rev4:default:prod` but still the same error.

yanfali, 2019-09-18:
@slipperydippery try this
`make planck/rev4:default:production`
I just typed that on my mac and it worked. You should end up with
```
File sizes:
   text	   data	    bss	    dec	    hex	filename
      0	  27916	      0	  27916	   6d0c	planck_rev4_default.hex
      0	   3944	      0	   3944	    f68	planck_rev4_default_bootloader.hex
      0	  31860	      0	  31860	   7c74	planck_rev4_default_production.hex
```
You want the production hex.

slipperydippery, 2019-09-18:
@yanfali  thanks again. I redownloaded the repo (I might have had an old one), and it worked. Sorry for the bother. Going to flash it to the board now

yanfali, 2019-09-18:
Feel free to close issue if you feel it has been resolved to your satisfaction.

slipperydippery, 2019-09-18:
@yanfali Thanks, it worked perfectly!

closed 2019-09-18 by slipperydippery