qmk/qmk_firmware #6766: Allow custom programming executables

tzarc, 2019-09-19:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

With dfu-util, the current release version doesn't support STM32L412 devices, as there's a timing change with the way that chip does DFU.

The master branch of dfu-util has a fix for this, but unfortunately nobody knows when a new release build will get made, given that the last one was in 2016.

Would be great if there could be an option to specify alternate executables when invoking things like dfu-util and avrdude etc. -- would allow for custom builds to be run locally without requiring a full release.

ReVanTis, 2019-09-20:
I think for Linux it is defined with these variables, or at least for WSL:
https://github.com/qmk/qmk_firmware/blob/master/util/activate_wsl.sh#L7
Not sure if it will work you override these variables.

drashna, 2019-09-26:
Actually, we do allow this already. 

https://github.com/qmk/qmk_firmware/blob/master/tmk_core/chibios.mk#L210

You just need to specify the executable when compiling (or in your rules.mk).

Just add:
```make
DFU_UTIL=/path/to/dfu_util_custom
```

sattaa, 2019-10-06:
Satta Matka rn.com

closed 2021-09-29 by tzarc