qmk/qmk_firmware #6780: Planck_EZ not compiling

Yatreyu, 2019-09-21:
Probably a total noob error but I have the following keymap:

`const` uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
	[0] = LAYOUT_planck_1x2uC(KC_QUOT, KC_COMM, KC_DOT, KC_P, KC_Y, LALT_T(KC_TAB), LCTL_T(KC_NLCK), KC_F, KC_G, KC_C, KC_R, KC_L, KC_A, KC_O, KC_E, KC_U, KC_I, KC_UP, LGUI(KC_CAPS), KC_D, KC_H, KC_T, KC_N, KC_S, KC_COLN, KC_J, KC_X, OSL(2), KC_LEFT, KC_DOWN, KC_RGHT, KC_K, KC_BSPC, KC_M, KC_W, KC_V, LT(1,KC_ESC), KC_Q, KC_Z, KC_QUES, LT(1,KC_SPC), KC_LEAD, LT(0,KC_SPC), LT(3,KC_NO), LT(6,KC_NO), KC_B, LT(4,KC_ENT)),
	[1] = LAYOUT_planck_1x2uC(KC_7, KC_8, KC_9, KC_PLUS, KC_PCMM, KC_TRNS, KC_TRNS, KC_NLCK, KC_PPLS, KC_P7, KC_P8, KC_P9, KC_4, KC_5, KC_6, KC_MINS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_PMNS, KC_P4, KC_P5, KC_P6, KC_1, KC_2, KC_3, KC_ASTR, KC_EQL, KC_TRNS, KC_TRNS, KC_PEQL, KC_PAST, KC_P1, KC_P2, KC_P3, KC_DOT, KC_0, KC_EQL, KC_SLSH, LSFT_T(KC_SPC), KC_TRNS, LT(0,KC_SPC), KC_PSLS, KC_PDOT, KC_P0, KC_PENT),
	[2] = LAYOUT_planck_1x2uC(KC_DQUO, KC_LT, KC_GT, LSFT(KC_P), LSFT(KC_Y), KC_TRNS, KC_TRNS, LSFT(KC_F), LSFT(KC_G), LSFT(KC_C), LSFT(KC_R), LSFT(KC_L), LSFT(KC_A), LSFT(KC_O), LSFT(KC_E), LSFT(KC_U), LSFT(KC_I), KC_TRNS, KC_TRNS, LSFT(KC_D), LSFT(KC_H), LSFT(KC_T), LSFT(KC_N), LSFT(KC_S), KC_SCLN, LSFT(KC_J), LSFT(KC_X), KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, LSFT(KC_K), KC_TRNS, LSFT(KC_M), LSFT(KC_W), LSFT(KC_V), KC_ESC, LSFT(KC_Q), LSFT(KC_Z), KC_EXLM, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, LSFT(KC_B), KC_TRNS),
	[3] = LAYOUT_planck_1x2uC(KC_CIRC, KC_LT, KC_GT, KC_BSLS, KC_SLSH, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_F7, KC_F8, KC_F9, KC_PIPE, KC_LPRN, KC_RPRN, KC_AT, KC_DLR, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_F4, KC_F5, KC_F6, KC_UNDS, KC_LBRC, KC_RBRC, KC_EQL, KC_HASH, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_F1, KC_F2, KC_F3, KC_TRNS, KC_LCBR, KC_RCBR, KC_TILD, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_F10, KC_TRNS),
	[4] = LAYOUT_planck_1x2uC(KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, TO(0), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, STN_S1, STN_TL, STN_PL, STN_HL, STN_ST1, KC_NO, STN_ST3, STN_FR, STN_LR, STN_TR, STN_TR, STN_DR, STN_S2, STN_KL, STN_WL, STN_RL, STN_ST2, KC_NO, STN_ST4, STN_RR, STN_BR, STN_SR, STN_SR, STN_ZR, KC_NO, KC_NO, KC_NO, STN_A, STN_O, KC_TRNS, STN_E, STN_U, KC_NO, KC_NO, KC_NO),
	[5] = LAYOUT_planck_1x2uC(KC_F, KC_G, KC_C, KC_R, KC_L, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_D, KC_H, KC_T, KC_N, KC_S, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_K, KC_BSPC, KC_M, KC_V, KC_W, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_LEAD, KC_NO, KC_B, KC_ENT, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO),
	[6] = LAYOUT_planck_1x2uC(KC_QUOT, KC_COMM, KC_DOT, KC_P, KC_Y, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_A, KC_O, KC_E, KC_U, KC_I, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_SCLN, KC_J, KC_X, OSM(MOD_LSFT), TO(0), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, MO(5), KC_Q, KC_Z, KC_QUES, KC_SPC, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO)
};`

And I understand trying to have a STENO layer is a complicating factor, but even when I try with just layers 0 to 2 I get the following error(s):

`$ make planck_ez:elephantshell
QMK Firmware 0.7.13
Making planck_ez with keymap elephantshell

arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/planck_ez/keymaps/elephantshell/keymap.c                                      keyboards/planck_ez/keymaps/elephantshell/keymap.c:5:8: error: implicit declaration of function 'LAYOUT_planck_1x2uC' [-Werror=implicit-function-declaration]
  [0] = LAYOUT_planck_1x2uC(KC_QUOT, KC_COMM, KC_DOT, KC_P, KC_Y, LALT_T(KC_TAB), LCTL_T(KC_NLCK), KC_F, KC_G, KC_C, KC_R, KC_L, KC_A, KC_O, KC_E, KC_U, KC_I, KC_UP, LGUI(KC_CAPS), KC_D, KC_H, KC_T, KC_N, KC_S, KC_COLN, KC_J, KC_X, OSL(2), KC_LEFT, KC_DOWN, KC_RGHT, KC_K, KC_BSPC, KC_M, KC_W, KC_V, LT(1,KC_ESC), KC_Q, KC_Z, KC_QUES, LT(1,KC_SPC), KC_LEAD, LT(0,KC_SPC), LT(3,KC_NO), LT(6,KC_NO), KC_B, LT(4,KC_ENT)),
        ^~~~~~~~~~~~~~~~~~~
keyboards/planck_ez/keymaps/elephantshell/keymap.c:5:8: error: initializer element is not constant
keyboards/planck_ez/keymaps/elephantshell/keymap.c:5:8: note: (near initialization for 'keymaps[0][0][0]')
keyboards/planck_ez/keymaps/elephantshell/keymap.c:6:8: error: initializer element is not constant
  [1] = LAYOUT_planck_1x2uC(KC_7, KC_8, KC_9, KC_PLUS, KC_PCMM, KC_TRNS, KC_TRNS, KC_NLCK, KC_PPLS, KC_P7, KC_P8, KC_P9, KC_4, KC_5, KC_6, KC_MINS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_PMNS, KC_P4, KC_P5, KC_P6, KC_1, KC_2, KC_3, KC_ASTR, KC_EQL, KC_TRNS, KC_TRNS, KC_PEQL, KC_PAST, KC_P1, KC_P2, KC_P3, KC_DOT, KC_0, KC_EQL, KC_SLSH, LSFT_T(KC_SPC), KC_TRNS, LT(0,KC_SPC), KC_PSLS, KC_PDOT, KC_P0, KC_PENT),
        ^~~~~~~~~~~~~~~~~~~
keyboards/planck_ez/keymaps/elephantshell/keymap.c:6:8: note: (near initialization for 'keymaps[1][0][0]')
keyboards/planck_ez/keymaps/elephantshell/keymap.c:7:8: error: initializer element is not constant
  [2] = LAYOUT_planck_1x2uC(KC_DQUO, KC_LT, KC_GT, LSFT(KC_P), LSFT(KC_Y), KC_TRNS, KC_TRNS, LSFT(KC_F), LSFT(KC_G), LSFT(KC_C), LSFT(KC_R), LSFT(KC_L), LSFT(KC_A), LSFT(KC_O), LSFT(KC_E), LSFT(KC_U), LSFT(KC_I), KC_TRNS, KC_TRNS, LSFT(KC_D), LSFT(KC_H), LSFT(KC_T), LSFT(KC_N), LSFT(KC_S), KC_SCLN, LSFT(KC_J), LSFT(KC_X), KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, LSFT(KC_K), KC_TRNS, LSFT(KC_M), LSFT(KC_W), LSFT(KC_V), KC_ESC, LSFT(KC_Q), LSFT(KC_Z), KC_EXLM, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, LSFT(KC_B), KC_TRNS),
        ^~~~~~~~~~~~~~~~~~~
keyboards/planck_ez/keymaps/elephantshell/keymap.c:7:8: note: (near initialization for 'keymaps[2][0][0]')
keyboards/planck_ez/keymaps/elephantshell/keymap.c:8:8: error: initializer element is not constant
  [3] = LAYOUT_planck_1x2uC(KC_CIRC, KC_LT, KC_GT, KC_BSLS, KC_SLSH, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_F7, KC_F8, KC_F9, KC_PIPE, KC_LPRN, KC_RPRN, KC_AT, KC_DLR, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_F4, KC_F5, KC_F6, KC_UNDS, KC_LBRC, KC_RBRC, KC_EQL, KC_HASH, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_F1, KC_F2, KC_F3, KC_TRNS, KC_LCBR, KC_RCBR, KC_TILD, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_F10, KC_TRNS),
        ^~~~~~~~~~~~~~~~~~~
keyboards/planck_ez/keymaps/elephantshell/keymap.c:8:8: note: (near initialization for 'keymaps[3][0][0]')
keyboards/planck_ez/keymaps/elephantshell/keymap.c:9:8: error: initializer element is not constant
  [4] = LAYOUT_planck_1x2uC(KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, TO(0), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, STN_S1, STN_TL, STN_PL, STN_HL, STN_ST1, KC_NO, STN_ST3, STN_FR, STN_LR, STN_TR, STN_TR, STN_DR, STN_S2, STN_KL, STN_WL, STN_RL, STN_ST2, KC_NO, STN_ST4, STN_RR, STN_BR, STN_SR, STN_SR, STN_ZR, KC_NO, KC_NO, KC_NO, STN_A, STN_O, KC_TRNS, STN_E, STN_U, KC_NO, KC_NO, KC_NO),
        ^~~~~~~~~~~~~~~~~~~
keyboards/planck_ez/keymaps/elephantshell/keymap.c:9:8: note: (near initialization for 'keymaps[4][0][0]')
keyboards/planck_ez/keymaps/elephantshell/keymap.c:10:8: error: initializer element is not constant
  [5] = LAYOUT_planck_1x2uC(KC_F, KC_G, KC_C, KC_R, KC_L, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_D, KC_H, KC_T, KC_N, KC_S, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_K, KC_BSPC, KC_M, KC_V, KC_W, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_LEAD, KC_NO, KC_B, KC_ENT, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO),
        ^~~~~~~~~~~~~~~~~~~
keyboards/planck_ez/keymaps/elephantshell/keymap.c:10:8: note: (near initialization for 'keymaps[5][0][0]')
keyboards/planck_ez/keymaps/elephantshell/keymap.c:11:8: error: initializer element is not constant
  [6] = LAYOUT_planck_1x2uC(KC_QUOT, KC_COMM, KC_DOT, KC_P, KC_Y, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_A, KC_O, KC_E, KC_U, KC_I, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_SCLN, KC_J, KC_X, OSM(MOD_LSFT), TO(0), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, MO(5), KC_Q, KC_Z, KC_QUES, KC_SPC, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO)
        ^~~~~~~~~~~~~~~~~~~
keyboards/planck_ez/keymaps/elephantshell/keymap.c:11:8: note: (near initialization for 'keymaps[6][0][0]')
cc1.exe: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:380: .build/obj_planck_ez_elephantshell/keyboards/planck_ez/keymaps/elephantshell/keymap.o] Error 1
Make finished with errors
make: *** [Makefile:561: planck_ez:elephantshell] Error 1`

If anyone has any suggestions that would be GREATLY appreciated. Looking forward to using my keyboard.

fauxpark, 2019-09-21:
I'm not sure what the *exact* problem is here, but I suspect your repo may be out of date. The Planck EZ has always been located in `keyboards/planck/ez` so you might be working off an old copypasted codebase.

drashna, 2019-09-21:
Yeah, I can confirm that it compiles correctly.   But there was an issue with it a while ago.   So, it sounds like you need to update your repo.

drashna, 2019-09-26:
@Yatreyu were you able to get this compiling?

Yatreyu, 2019-09-26:
No I'm having troubles figuring it out. I ran the msys2 install
instructions in the qmkdocs, the practice clueboard66/rev3 compiles.

But still getting error messages.

Is it ok to post my directory on GitHub?

noroadsleft, 2019-09-26:
> No I'm having troubles figuring it out. I ran the msys2 install instructions in the qmkdocs, the practice clueboard66/rev3 compiles. But still getting error messages. Is it ok to post my directory on GitHub?

@Yatreyu,

I can see your fork at https://github.com/Yatreyu/qmk_firmware/tree/master, and it is well out of date.

Open up MSYS2, cd to wherever you have your `qmk_firmware` folder on your computer, and run `git remote -v`. It should return something like:

```
$ git remote -v
origin  https://github.com/Yatreyu/qmk_firmware.git (fetch)
origin  https://github.com/Yatreyu/qmk_firmware.git (push)
```
(The `origin` part might be named something else.)

If the only remote repository is your own fork, you should add QMK's repository as a remote with:

```
git remote add upstream https://github.com/qmk/qmk_firmware.git
```

Once you've done that, `git remote -v` should show both your fork and QMK's repository. Run the following commands, one command per line:

```
git checkout master
git fetch upstream
git pull upstream master
git push origin master
```

This will update the master branch of your fork on both your computer and GitHub, so they're all exactly the same.

Once you've done this, you should be able to compile Planck EZ firmware with `make planck/ez:default` (note the `/`), or your own keymap with `make planck/ez:elephantshell` (assuming it doesn't have other errors).

Yatreyu, 2019-09-30:
```
$ git remote -v
origin  https://github.com/qmk/qmk_firmware.git (fetch)
origin  https://github.com/qmk/qmk_firmware.git (push)
```
Is what I got so I did this: 
```
$ git remote -v
downstream      https://github.com/Yatreyu/qmk_firmware.git (fetch)
downstream      https://github.com/Yatreyu/qmk_firmware.git (push)
origin  https://github.com/qmk/qmk_firmware.git (fetch)
origin  https://github.com/qmk/qmk_firmware.git (push)
upstream        https://github.com/qmk/qmk_firmware.git (fetch)
upstream        https://github.com/qmk/qmk_firmware.git (push)
```

Then when I try and complete Elephantshell:

```
$ make planck/ez:elephantshell
QMK Firmware 0.7.13
Making planck/ez with keymap elephantshell

arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/planck/ez/keymaps/elephantshell/keymap.c                                      In file included from keyboards/planck/ez/keymaps/elephantshell/keymap.c:2:0:
quantum/keymap_extras/keymap_steno.h:25:16: error: 'QK_STENO' undeclared here (not in a function)
     STN__MIN = QK_STENO,
                ^~~~~~~~
In file included from keyboards/planck/planck.h:9:0,
                 from keyboards/planck/ez/keymaps/elephantshell/keymap.c:1:
keyboards/planck/ez/keymaps/elephantshell/keymap.c:5:347: error: 'KC_LEAD' undeclared here (not in a function)
  [0] = LAYOUT_planck_mit(KC_QUOT, KC_COMM, KC_DOT, KC_P, KC_Y, LALT_T(KC_TAB), LCTL_T(KC_NLCK), KC_F, KC_G, KC_C, KC_R, KC_L, KC_A, KC_O, KC_E, KC_U, KC_I, KC_UP, LGUI(KC_CAPS), KC_D, KC_H, KC_T, KC_N, KC_S, KC_COLN, KC_J, KC_X, OSL(2), KC_LEFT, KC_DOWN, KC_RGHT, KC_K, KC_BSPC, KC_M, KC_W, KC_V, LT(1,KC_ESC), KC_Q, KC_Z, KC_QUES, LT(1,KC_SPC), KC_LEAD, LT(0,KC_SPC), LT(3,KC_NO), LT(6,KC_NO), KC_B, LT(4,KC_ENT)),
                                                                                                                                                                                                                                                                                                                                                           ^
keyboards/planck/ez/ez.h:34:32: note: in definition of macro 'LAYOUT_planck_1x2uC'
     { k37, k38, k39, k33, k34, k35   } \
                                ^~~
keyboards/planck/ez/keymaps/elephantshell/keymap.c:5:8: note: in expansion of macro 'LAYOUT_planck_mit'
  [0] = LAYOUT_planck_mit(KC_QUOT, KC_COMM, KC_DOT, KC_P, KC_Y, LALT_T(KC_TAB), LCTL_T(KC_NLCK), KC_F, KC_G, KC_C, KC_R, KC_L, KC_A, KC_O, KC_E, KC_U, KC_I, KC_UP, LGUI(KC_CAPS), KC_D, KC_H, KC_T, KC_N, KC_S, KC_COLN, KC_J, KC_X, OSL(2), KC_LEFT, KC_DOWN, KC_RGHT, KC_K, KC_BSPC, KC_M, KC_W, KC_V, LT(1,KC_ESC), KC_Q, KC_Z, KC_QUES, LT(1,KC_SPC), KC_LEAD, LT(0,KC_SPC), LT(3,KC_NO), LT(6,KC_NO), KC_B, LT(4,KC_ENT)),
        ^~~~~~~~~~~~~~~~~
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:380: .build/obj_planck_ez_elephantshell/keyboards/planck/ez/keymaps/elephantshell/keymap.o] Error 1
Make finished with errors
make: *** [Makefile:561: planck/ez:elephantshell] Error 1
```

So it is still not getting the job done :(

fauxpark, 2019-09-30:
That looks like you haven't added `STENO_ENABLE = yes` and `LEADER_ENABLE = yes` in your rules.mk. The things it's complaining are undeclared are dependent on those features being enabled.

Yatreyu, 2019-09-30:
Hey, thanks for the ongoing troubleshooting to get this working. 

My Rules.mk looks like it has those enabled: 

```
# Cortex version
MCU  = STM32F303

# Interrupt driven control endpoint task(+60)
OPT_DEFS += -DINTERRUPT_CONTROL_ENDPOINT

# Build Options
#   change to "no" to disable the options, or define them in the Makefile in
#   the appropriate keymap folder that will get included automatically
#
BOOTMAGIC_ENABLE = no      # Virtual DIP switch configuration(+1000)
## (Note that for BOOTMAGIC on Teensy LC you have to use a custom .ld script.)
MOUSEKEY_ENABLE = no      # Mouse keys(+4700)
EXTRAKEY_ENABLE = yes       # Audio control and System control(+450)
CONSOLE_ENABLE = no         # Console for debug(+400)
COMMAND_ENABLE = no       # Commands for debug and configuration
NKRO_ENABLE = yes            # Nkey Rollover - if this doesn't work, see here: https://github.com/tmk/tmk_keyboard/wiki/FAQ#nkro-doesnt-work
BACKLIGHT_ENABLE = no      # Enable keyboard backlight functionality
MIDI_ENABLE = no            # MIDI controls
AUDIO_ENABLE = yes           # Audio output on port C6
UNICODE_ENABLE = yes         # Unicode
BLUETOOTH_ENABLE = no       # Enable Bluetooth with the Adafruit EZ-Key HID
RGBLIGHT_ENABLE = no        # Enable WS2812 RGB underlight.
API_SYSEX_ENABLE = no

# Do not enable SLEEP_LED_ENABLE. it uses the same timer as BACKLIGHT_ENABLE
SLEEP_LED_ENABLE = no    # Breathing sleep LED during USB suspend
#SLEEP_LED_ENABLE = yes  # Breathing sleep LED during USB suspend

CUSTOM_MATRIX = no # Custom matrix file
# SERIAL_LINK_ENABLE = yes
ENCODER_ENABLE = yes
RGB_MATRIX_ENABLE = IS31FL3737

LAYOUTS = ortho_4x12 planck_mit
LAYOUTS_HAS_RGB = yes

LEADER_ENABLE = yes
COMBO_ENABLE = yes
FORCE_NKRO = yes
VIRTSER_ENABLE = yes
STENO_ENABLE = yes
```

I was wondering if it was in the right directory. I have it here: 

[Directory](https://imgur.com/a/CGeDktO)

I thought I might just outline if I have a slippery slope of errors by showing what I want my leader key to do: 

```
// Full keymap to try once I get just the keys working
#include QMK_KEYBOARD_H
#include "keymap_steno.h"

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
	[0] = LAYOUT_planck_1x2uC(KC_QUOT, KC_COMM, KC_DOT, KC_P, KC_Y, LALT_T(KC_TAB), LCTL_T(KC_NLCK), KC_F, KC_G, KC_C, KC_R, KC_L, KC_A, KC_O, KC_E, KC_U, KC_I, KC_UP, LGUI(KC_CAPS), KC_D, KC_H, KC_T, KC_N, KC_S, KC_COLN, KC_J, KC_X, OSL(2), KC_LEFT, KC_DOWN, KC_RGHT, KC_K, KC_BSPC, KC_M, KC_W, KC_V, LT(1,KC_ESC), KC_Q, KC_Z, KC_QUES, LT(1,KC_SPC), KC_LEAD, LT(0,KC_SPC), LT(3,KC_NO), LT(6,KC_NO), KC_B, LT(4,KC_ENT)),
	[1] = LAYOUT_planck_1x2uC(KC_7, KC_8, KC_9, KC_PLUS, KC_PCMM, KC_TRNS, KC_TRNS, KC_NLCK, KC_PPLS, KC_P7, KC_P8, KC_P9, KC_4, KC_5, KC_6, KC_MINS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_PMNS, KC_P4, KC_P5, KC_P6, KC_1, KC_2, KC_3, KC_ASTR, KC_EQL, KC_TRNS, KC_TRNS, KC_PEQL, KC_PAST, KC_P1, KC_P2, KC_P3, KC_DOT, KC_0, KC_EQL, KC_SLSH, LSFT_T(KC_SPC), KC_TRNS, LT(0,KC_SPC), KC_PSLS, KC_PDOT, KC_P0, KC_PENT),
	[2] = LAYOUT_planck_1x2uC(KC_DQUO, KC_LT, KC_GT, LSFT(KC_P), LSFT(KC_Y), KC_TRNS, KC_TRNS, LSFT(KC_F), LSFT(KC_G), LSFT(KC_C), LSFT(KC_R), LSFT(KC_L), LSFT(KC_A), LSFT(KC_O), LSFT(KC_E), LSFT(KC_U), LSFT(KC_I), KC_TRNS, KC_TRNS, LSFT(KC_D), LSFT(KC_H), LSFT(KC_T), LSFT(KC_N), LSFT(KC_S), KC_SCLN, LSFT(KC_J), LSFT(KC_X), KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, LSFT(KC_K), KC_TRNS, LSFT(KC_M), LSFT(KC_W), LSFT(KC_V), KC_ESC, LSFT(KC_Q), LSFT(KC_Z), KC_EXLM, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, LSFT(KC_B), KC_TRNS),
	[3] = LAYOUT_planck_1x2uC(KC_CIRC, KC_LT, KC_GT, KC_BSLS, KC_SLSH, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_F7, KC_F8, KC_F9, KC_PIPE, KC_LPRN, KC_RPRN, KC_AT, KC_DLR, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_F4, KC_F5, KC_F6, KC_UNDS, KC_LBRC, KC_RBRC, KC_EQL, KC_HASH, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_F1, KC_F2, KC_F3, KC_TRNS, KC_LCBR, KC_RCBR, KC_TILD, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_F10, KC_TRNS),
	[4] = LAYOUT_planck_1x2uC(KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, TO(0), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, STN_S1, STN_TL, STN_PL, STN_HL, STN_ST1, KC_NO, STN_ST3, STN_FR, STN_LR, STN_TR, STN_TR, STN_DR, STN_S2, STN_KL, STN_WL, STN_RL, STN_ST2, KC_NO, STN_ST4, STN_RR, STN_BR, STN_SR, STN_SR, STN_ZR, KC_NO, KC_NO, KC_NO, STN_A, STN_O, KC_TRNS, STN_E, STN_U, KC_NO, KC_NO, KC_NO),
	[5] = LAYOUT_planck_1x2uC(KC_F, KC_G, KC_C, KC_R, KC_L, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_D, KC_H, KC_T, KC_N, KC_S, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_K, KC_BSPC, KC_M, KC_V, KC_W, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_LEAD, KC_NO, KC_B, KC_ENT, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO),
	[6] = LAYOUT_planck_1x2uC(KC_QUOT, KC_COMM, KC_DOT, KC_P, KC_Y, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_A, KC_O, KC_E, KC_U, KC_I, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_SCLN, KC_J, KC_X, OSM(MOD_LSFT), TO(0), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, MO(5), KC_Q, KC_Z, KC_QUES, KC_SPC, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO)
};

[... 207 lines cut ...]
    	unregister_code(KC_V);
    	unregister_code(KC_LCTL);
    	}
		//Cut function
        SEQ_ONE_KEY(KC_T)
    	{
    	register_code(KC_LCTL);
    	register_code(KC_X);
    	unregister_code(KC_X);
    	unregister_code(KC_LCTL);
    	}
    	//Alt-Tab function
        SEQ_TWO_KEYS(KC_K, KC_T)
    	{
    	register_code(KC_LALT);
    	register_code(KC_TAB);
    	unregister_code(KC_TAB);
        unregister_code(KC_RALT);
    }
}
```

I want to be at the hospital use my leader key magic on any of the workstations and get my patient notes done quickly. I don't know if they are formatted correctly. 

This is not the keymap I'm trying to compile now. Just the layers as per my original post. Just fishing for some expertise in leader keys or macros in general as well as being really eager to get my basic key layout working.

fauxpark, 2019-09-30:
You seem to be looking at the wrong directory. You're compiling the `planck/ez` keyboard, which is correct, but those screenshots are of a `planck_ez` folder, which does not exist in master.

closed 2020-06-13 by stale[bot]