qmk/qmk_firmware #6813: Encoder on slave side doesn't report accurately

LouWii, 2019-09-25:
## Describe the Bug

I have a split keyboard with a ProMicro on each side. I've tested a few features of QMK and everything works fine.

I've been playing with a rotary encoder and it works perfectly on the master side, but the one on the slave side doesn't report accurately. When I rotate it, the action only triggers from time to time, randomly. My encoder has ticks or clicks, or whatever that's called. On the master side, each click triggers the action properly. On the slave side, it's after 2 clicks, or 3, or 7. It's totally random.

## System Information

 - Keyboard:  Ergomax
   - Revision (if applicable):  rev1
 - Operating system: Windows 10
 
Source code in my repo: https://github.com/LouWii/qmk_firmware/tree/ergomax-keyboard/keyboards/ergomax/keymaps/etchasketch

## Additional Context

I've plugged in the slave side directly to make it act as master. The encoder works fine in that case, so I really think it's the issue only happens when the encoder is on the slave side.

No idea where to start debugging this.

drashna, 2019-09-26:
Could you update your repo, there are some recent changes that may change/fix/break the behavior for the encoders.

LouWii, 2019-09-28:
I've pullled from master, merged in my branch, recompiled, flashed both sides. Unfortunately, the problem is still present.

drashna, 2019-10-02:
Could you try reverting this commit:
46c49ae4e639509ceb6e81c460855d2eb89086b8

LouWii, 2019-10-20:
Hi!

Sorry for the late response. I finally had time to test this and unfortunately, it didn't change anything. The behaviour is the same.

closed 2022-08-29 by zvecr