qmk/qmk_firmware #6824: qmk compile could be context sensitive about keyboard directory

yanfali, 2019-09-26:
Idea is: you are in `keyboards/keebio/iris/rev3` and you type `qmk compile` if no defaults are set it detects the directory you are in and compiles that with the default keymap. If you have a default keymap set in your global config it attempts to compile your personal keymap if it exists.

This would make the qmk compile context sensitive to where it is being run from to source parameters for keyboard and keymap.

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Think of these as fall back sources for parameters when there are none supplied by the defaults or the cli.

skullydazed, 2019-09-26:
I think a good argument could be made for CWD overriding configuration. It would be consistent with how the global CLI operates- configuration can specify the default `qmk_firmware` location but your current directory will override that configuration.

cfbender, 2019-09-26:
I'd love to take this one for hacktoberfest if that's alright! I think I should be able to do it, thought it will be some of the first python I've written for a bit. If you'd prefer someone more experienced I'd understand!

ThiagoWhispher, 2019-10-07:
This would be very interesting for users who use it, leaving the complexity to QMK.

vaibhavbisht06, 2019-10-22:
Is there any way I can add images to this guide

sakshamtaneja21, 2019-10-26:
for this ill open a pr and then can check @admin

DavidYangDev, 2019-10-27:
@vaibhavbisht06  Where would you want to add images? just confused because I'm not really great at this either

rishav142k, 2019-10-27:
Is anyone working on this issue because I would like to take it.

PR #6884 (2020-02-07): Add QMK Compile Context Sensitivity

## Description

Adds context sensitivity to `qmk compile`. Allows the command to be run within a keyboard directory, given no other CLI arguments, and will compile the default keymap for the given keyboard.

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR
 
* Closes #6824 

## Checklist

- [x] My code follows the code style of this project.
- [x] My change requires a ch [... cut]
closed 2020-02-07 by Erovia