qmk/qmk_firmware #6827: dz65rgb led matrix - bug or faulty pcb?

thorgnir, 2019-09-26:
When `rgb_matrix_set_color(8, 0xFF, 0xFF, 0xFF);` is called - it lights led under capslock with white color as intended. But also it lights led under "u" with red.
May be my pcb is faulty, I don't know. 
Is it possible that there is some bug in led matrix definition?

thorgnir, 2019-09-26:
I just found that "u" key led has index 0, so issue can be somehow related to this

drashna, 2019-10-02:
Try using a solid color for the whole keyboard.

My guess is that that the led under "U" may be bad.  It may need to be fixed.

thorgnir, 2019-10-02:
Solid color works ok (i.e. green or white backlight on whole keyboard - led under "U" is also green or white). If I tap caps lock and call rgb_matrix_set_color(8, 0xFF, 0xFF, 0xFF); - led under "U" goes to pink (if before that it was white) or yellow (if before that it was green).

elbaidthe1st, 2020-05-26:
Did you ever fix this? I’m having the exact same problem

closed 2022-08-28 by zvecr