qmk/qmk_firmware #6829: Host-controlled colors

xylo04, 2019-09-27:
Create a protocol (and ideally, a low-level host library/language bindings) for the computer to send color layouts, and implement the color layout in firmware.

## Feature Request Type

- [ ] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

Continuing off of [r/ErgoDoxEZ](https://www.reddit.com/r/ErgoDoxEZ/comments/d9l4nq/computercontrolled_colors/?utm_source=share&utm_medium=web2x) and [r/olkb](https://www.reddit.com/r/olkb/comments/d9lqbd/computercontrolled_colors_in_qmk/?utm_source=share&utm_medium=web2x): I'd like to set up a program on the computer that controls the colors. I would personally set this up to display continuous integration status, but there are plenty of potential use cases. As a point of reference, see Razer Chroma.

In the Reddit discussion, @drashna said the overall idea is possible and that the Raw HID protocol would support ferrying the data, but that the data structures and protocol would need to be formalized, and then both the host and firmware ends would need to be implemented.

I think this idea was referenced in passing in https://github.com/qmk/qmk_firmware/issues/5934#issuecomment-494029312 mentioning "bidirectional HID comms" for colors. If I understood correctly, that internal color API change would ease doing the host communication bit.

I'll probably play with this a bit, but firmware is outside my comfort zone.

ReVanTis, 2019-09-29:
Many have implemented their own protocol through HID raw messages.

I've built my keymap and command line tool here:
https://github.com/ReVanTis/xd75_qmk/blob/master/keymap.c#L441-L456
https://github.com/ReVanTis/XD75RawHid/blob/master/Program.cs#L212-L236

All the credits go to Zeal60 fork from Wilba6582:
https://github.com/Wilba6582/zeal60
https://github.com/Wilba6582/qmk_firmware/tree/zeal60

I also agree at this point it seems possible we can try to make some official protocol, maybe not only for color but for all the bi-directional HID communications based features.
(key press report for statistics? Keymap/layer automatic switching based on host application?)

yanfali, 2019-09-30:
https://github.com/antonpup/Aurora

drashna, 2019-10-02:
You can use both Raw HID (which is what VIA uses) or you could use Virtual Serial to open bidirectional communication with the host system.

The problem with both, is that you're basically implementing the protocol manually.  The zeal boards use VIA and may be a good starting point.

closed 2020-03-30 by stale[bot]

xylo04, 2026-08-16:
I'm prototyping this host color idea again for my own use. If that goes anywhere, the ultimate solution will probably need to sit behind https://github.com/qmk/qmk_firmware/issues/11567.