qmk/qmk_firmware #6846: MIDI device isn't working on win10

hsgw, 2019-09-30:
## Describe the Bug
MIDI device can't be recognized correctly on windows10 64bit.
I revert commit before changing lib/lufa to submodule, it works.

## Additional Context
There is no probrem on OSX, ubuntu, and win7.

drashna, 2019-10-02:
If that's the case, one of the files in `tmk_core/protocol/usb_descriptors.c` wasn't probably changed when LUFA was changed over to a submodule.

LouWii, 2019-11-23:
@hsgw Have you tried that fix: https://github.com/qmk/qmk_firmware/pull/7258 ? It's been merged in.

drashna, 2019-12-01:
@hsgw have you had a chance to update the repo and test to see if this is still an issue?

tzarc, 2022-08-03:
Hasn't been updated in 2.5 years, going to assume this is no longer an issue.

closed 2022-08-03 by tzarc