qmk/qmk_firmware #6851: Flexible Keymap Layouts

Micro-Biology, 2019-10-01:
## Flexible keymap layouts

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ *] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

The ability to define layouts more flexibly. Possibly by row. I am going to use the DZ60 as an example because I am most familiar with that board, and the dz60.h file is a perfect example of why a feature like this would be great.

Most of the layouts of keyboards only differ by a few things for example the dz60 has the option to have split back space or 2u, different bottom rows, different enters, and different 4th row.

On their own these amount to 2-4 options per row (2 rows for the enter rows due to iso using both) however together all the combinations amount to a lot of different keymaps making it confusing.

drashna, 2019-10-01:
The problem with that, is that it's very hard to support all possible layouts.  

Specifically, to do so, they'd need to be supported in the keyboard's header file. But some boards, like the dz60 could have hundreds of possible permutations, if not more.   Supporting every one of them is ... basically impossible. 

For that reason, we generally have "`LAYOUT_all` defined for the boards. This is a layout that has every possible position enabled, even if you can't physically use all of them (eg, blocked by other keys, not enough room, etc).

Micro-Biology, 2019-10-01:
> The problem with that, is that it's very hard to support all possible layouts.
>
> Specifically, to do so, they'd need to be supported in the keyboard's header file. But some boards, like the dz60 [...]

Thanks for fixing my tick box.

Personally I think just having LAYOUT_all is fine, and I agree having every possible combination is a bad idea, it is very messy.
I would to go clean up the dz60 page but that would involve either:

- Individually changing everyones keymaps to support a handful of layouts, not ideal as that would be incredibly time consuming
- Purging (almost) all the keymaps and reducing the amount of layouts to a few that made sense with no overlapping layouts which are basically the same, again not ideal as people would have to adjust to the fewer but all encompassing layouts.

I was more wondering if it was possible to just select which options you are using and it work out your keymap for you? Or maybe you have a separate keymap per row? Whilst it would still require editing everyones keymaps it would also prevent the need to fill in KC_NO everywhere where keys weren't present.

closed 2020-01-29 by stale[bot]