qmk/qmk_firmware #6856: Is QMK compatible with optical encoder?

HMarxen, 2019-10-01:
The documentation says to connect the common pin of an encoder to ground but all optical encoders that I’ve found seem to connect their optical transistors to VCC. 
https://static6.arrow.com/aropdfconversion/b37bee95f5eae7e7f00cc37d7f45757e265dbb6b/50924574275363em14.pdf
Would that still be compatible without messing with extra transistors? Do I need to change the Firmware? In Arduino I’d know that I just had to change HIGH to LOW somewhere but the syntax of proper C has me a bit confused. 
Also, I can't really test this. Since this is a quite pricy component, I wanted to ask before buying.

drashna, 2019-10-02:
Well, we have support for rotary encoders:
https://docs.qmk.fm/#/feature_encoders

A quick look and this may work the same? 

Do you have a sample for arduino code?

HMarxen, 2019-10-02:
Thank you for the reply. Yes, I know QMK supports encoders and I’ve looked at the code. But my point is that optical encoders work a bit differently than normal drag contact encoders. It uses an LED, two photo transistors and an interrupter wheel to generate the signal. With normal encoders you can connect the common pin of A and B to whatever you want but an optical encoder behaves as if the common pin is connected to VCC. And there is nothing I can change about it since everything is in a dust proof package. 
The benefit of optical encoders is that they don’t feel so scratchy and have a far, far longer lifespan because there is no friction on the conductive parts.
Right now QMK seems to use the internal pullup on A and B, so If it were written in Arduino I would be looking for something like:
pinMode(2, INPUT_PULLUP);
and change that to normal “INPUT”.

HMarxen, 2019-10-03:
Sorry, on a second reading I think you understood what I meant. ^^
I think I am slowly getting my head around this. I don’t think it would work the same. The optical encoder sends out a small voltage (up to 0.8V) on Low and around 4V on High. So, If I understand it correctly, the internal pullup connects to VCC, the encoder does too and the Firmware waits for the pin to go Low. So, nothing would ever happen. I think I need to add two Pulldowns on my end and change the Firmware so that it waits for the pin to go High.  
Would perhaps changing:
```
setPinInputHigh(encoders_pad_a[i]);
setPinInputHigh(encoders_pad_b[i]);
```
To:
```
setPinInputLow(encoders_pad_a[i]);
setPinInputLow(encoders_pad_b[i]);
```
On line 68/69 in encoder.c be enough?

yanfali, 2019-10-03:
You probably wouldn't change the values, you would just add a define around the existing behavior so that when you set ENABLE_OPTICAL_ENCODERS in your config.h it would use that code instead of the default. You'll have to test it out of course first with actual hardware before we would accept it.

HMarxen, 2019-10-03:
I will but I am far away from it yet.^^
So, would what I'm describing above theoretically work?

drashna, 2019-10-12:
If yan is right about that, then yeah, it should.  You'd just need to change the code in `/quantum/encoders.c` and test it out.

Daveyr, 2020-12-14:
Did you get anywhere with this? I'm having trouble with a rotary encoder that I think is an EC11 but I'm not sure. I followed the thread above to modify the encoder.c file but it is still unresponsive. I've also laid the whole thing out on a breadboard, swapped A and B lines, and changed the encoder resolution but nothing seems to work.

closed 2022-08-28 by zvecr