qmk/qmk_firmware #6866: Dc-2014 Print Screen button

SneakyRobb, 2019-10-02:
Hi,

I am having an issue where my print screen button on the DC-2014 keyboard.

It seems to give the code 37 which is Pad_asterisk when I press print screen. 

Could you add this additional layout where that key is changed to the layout options for 

keyboard "converter_xt_usb?"

I think my file might not be properly formated or correct though. I added a new file. I think I accidentally pasted over the shift key.

*I found this tmk documentation outlining what is happening as well.
https://github.com/tmk/tmk_keyboard/issues/552

Thanks,
Robb

[converter_xt_usb_dc2014b.txt](https://github.com/qmk/qmk_configurator/files/3677317/converter_xt_usb_dc2014b.txt)

noroadsleft, 2019-10-02:
@SneakyRobb I think I might be starting to wrap my head around this.

Because Print Screen and Pad Asterisk are on the same key on your keyboard, you want the key to send Print Screen when Num Lock is disable, but Pad Asterisk when Num Lock is enabled, is that correct?

SneakyRobb, 2019-10-04:
@noroadsleft 

I understand this, it really confused me. It is not the biggest issue, it is just a minor thing. I think I understand it now.  Really its just to change the layout on the main QMK thing a bit. 

Basically on the XT keyboard the Asterisk and PrintScreen were the same key. 

For whatever reason, in TMK, the asterisk key was put down on the XT layout, as "Print Screen."

But the key is asterisk first, you had to press shift, to print screen. 

So now in QMK, if I open the configurator, to the keyboard "converter/xt_usb"
and layout "LAYOUT_xt"
here:
https://config.qmk.fm/#/converter/xt_usb/LAYOUT_xt

The key under enter is shown as Print Screen. If I use the configurator to map this key, it wont work. Because that key is Numpad-Asterisk. ''

That is basically it.

noroadsleft, 2019-10-05:
So it's always Numpad Asterisk no matter what is assigned there? I'm still confused.

PR #7039 (2019-10-26): Fix Print Screen key for XT to USB converter

## Description

Following along with [this commit](https://github.com/tmk/tmk_keyboard/commit/594d7d85da35637b3757fc0948e007358b57c72f) from TMK.

The Print Screen key on the XT is accessible via the keypad `*` key (presumably with Shift), however the layout macro was wrong. When the scancode `37` is received, the converter should emit `KC_PAST`. When the scancode `E0 37` is received, the `move_e0code()` function in matrix.c is called, and should activate the matrix key at `K54` to produce `KC_PSCR`.

@SneakyRobb could you please test this?

## Typ [... cut]
closed 2019-10-26 by drashna