qmk/qmk_firmware #6872: Convert new_keyboard.sh to a python command

skullydazed, 2019-10-02:
## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Similar to #6066, we should convert new_keyboard.sh to a qmk command.

Some thoughts on implementation:

* The cleanest way to handle the templates is probably by using `.format()` string formatting. 
* Like new_keymap the user should be able to supply answers on the command line. Any information not provided should be prompted for.
* This is an instance where most (all?) CLI flags should be `arg_only=True`.

HBBisenieks, 2019-10-02:
Is the intention that this be, essentially, a drop-in replacement for the current .sh version with the addition of CLI flags that supersede the interactive prompts?

skullydazed, 2019-10-03:
As a start, yes. I think there will be room to iterate and improve on it once that is done, and I'd welcome most of those improvements as part of the initial PR. For example, support for arm keyboards.

HBBisenieks, 2019-10-03:
I'm making a branch on my fork to get the ball rolling on this, and I'll open a PR when it's at least feature-complete with the .sh version.

stjohnjohnson, 2019-10-03:
@HBBisenieks sorry!

HBBisenieks, 2019-10-03:
Hey no worries! Your implementation looks way better than mine was looking in the planning stages.

ThiagoWhispher, 2019-10-07:
Hi @skullydazed, what is the advantage of replacing .sh with a python command? Better debugging?

lf-, 2019-10-07:
@ThiagoWhispher I'm going to go for maintainability, consistency in output/behaviour, reliability, and moving all the shell stuff to the qmk utility.

stjohnjohnson, 2019-10-07:
I like the idea of centralizing all these commands into a single CLI.  Combine that with the Docker container and you have a single command with no dependencies (besides Docker) for all operations.

ThiagoWhispher, 2019-10-07:
@lf- now I understand how advantageous this is.
@stjohnjohnson I working with Docker, and strongly recommend, very practical to work.

helios1101, 2019-10-07:
Are these docker images hosted on Docker Hub too? Just wanted to start my journey in containers as it would be my first experience with dockers contribution.

stjohnjohnson, 2019-10-08:
@helios1101 https://hub.docker.com/r/qmkfm/qmk_firmware for example

sakshamtaneja21, 2019-10-26:
for this please tell me what exact changes you needed

DavidYangDev, 2019-10-27:
Looking at the other comments, at first I was confused about the purpose of it now, but I rethought about it and it's usage is quite widespread. Is there a PR that tackles this?

DavidYangDev, 2019-10-27:
Just wondering, is there a pr that has handles/is handing this request?

PR #12795 (2022-02-08): Initial pass at data driven new-keyboard subcommand

## Description

Builds on previous iterations by leaning on the community defaults and json schemas.

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #6872

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I [... cut]
closed 2022-08-03 by tzarc