qmk/qmk_firmware #6873: Add custom keycode support to JSON files

skullydazed, 2019-10-02:
## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

When a JSON file is compiled (`qmk compile configurator_export.json`) it's currently not possible to use custom keycodes. We should add a new key, `custom_keycodes`, which the user can populate.

For the initial implementation we should add this to the keyboard template:

```c
enum custom_keycodes {
__CUSTOM_KEYCODES_HERE__
}
```

We will then need to extend the `generate()` function in `lib/python/qmk/keymap.py` to also generate and write values for this enum.

This feature will not be immediately useful but other features (such as macros and interacting with userspace code) will build on this feature.

jorgemanzo, 2019-10-06:
Would this go in the `quantum/template/base/keymaps/default/keymap.c` template?

If so I think I see where `__CUSTOM_KEYCODES_HERE__` would be used in `keymap.py`. I think we would be adding a `keymap_c.replace('__CUSTOM_KEYCODES_HERE__', custom_keycodes)`? In a configurator export, are custom keycodes `ANY(<custom keycode>)`?

skullydazed, 2019-10-06:
It would go into keymap.py.

There is currently no way to specify a custom keycode in configurator exports. This issue is a step towards being able to specify custom keycodes in configurator.

jorgemanzo, 2019-10-06:
So just placing an enum in the default template? https://github.com/jorgemanzo/qmk_firmware/commit/3b451a17617cd22d835b2d7329faca09c8e94e7a

Are there any keymap.c files in the repo that define custom keycodes? I have a fair idea of how layers and keycodes are taken from the JSON in keymap.py/

skullydazed, 2019-10-06:
It should go above the layer definition but yes you're on the right track there. The next will be to extend a configurator export with `"custom_keycodes": ["CUSTOM1", "CUSTOM2"]` and then write the code to populate that.

The documentation for custom keycodes is here: https://docs.qmk.fm/#/custom_quantum_functions?id=custom-keycodes

DavidYangDev, 2019-10-27:
I'm currently confused about the process of doing this, is this already in the making?

jorgemanzo, 2019-10-27:
I think (might be wrong here) a good place to start would be to look in `lib/python/qmk/keymap.py`. At the top there should be a `__KEYMAP_GOES_HERE__` marker. This whole python string gets turned into a keymap.c file, and I think its done within `keymap.py`.

This python code will go through a configurator export JSON and extract out the keycodes and keymap and format them into the C `keymaps[]`.

zvecr, 2023-11-14:
Implemented in #19925

closed 2023-11-14 by zvecr