qmk/qmk_firmware #6880: CLI command to convert KLE to JSON

skullydazed, 2019-10-02:
## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

Currently you have to use the API (by way of a single use web page) to convert KLE to info.json. We should have a CLI command to do this as well.

The API entrypoint for the converter is here: https://github.com/qmk/qmk_api/blob/master/web.py#L209

cfbender, 2019-10-04:
Working on this now!

helios1101, 2019-10-10:
Just a suggestion, while I was exploring, as a beginner I personally felt that it would be great if the virtual environment would be made in the beginning so that the requirements don't hamper the rest of the environment.

skullydazed, 2019-10-10:
@helios1101 We have talked about doing that in the past (in fact if you install the qmk cli using homebrew it uses a virtualenv) but for right now are not doing so. Users can setup their own virtualenv if they feel it is important.

That said, we're out of scope for the current issue. If you'd like to discuss this further please create a new issue for the topic.

vaibhavbisht06, 2019-10-22:
Can any repo maintainer help me to understand the 2 flie of project

PR #6898 (2019-11-13): Add cli convert subcommand, from raw KLE to JSON
## Description

Adds a convert subcommand to convert raw KLE data files to configurator JSON.

## Types of Changes

- [x] Core
- [ ] Bugfix
- [x] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [x] Documentation

## Issues Fixed or Closed by This PR

* Resolves  #6880 

## Checklist

- [x] My code follows the code style of this project.
- [x] My change requires a change to the documentation.
- [x] I have updated the documentation accordingly.
- [x] I have read th [... cut]
closed 2019-11-13 by skullydazed