qmk/qmk_firmware #6890: CLI command to flash a keyboard

skullydazed, 2019-10-03:
## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

This command should behave similarly to `qmk compile`, except that it will call `make <keyboard>:<keymap>:flash` instead. It should also accept an argument to specify the flash command, with the available options corresponding to qmk's make options (`dfu`, 'dfu-util`, `teensy`, etc).

yanfali, 2019-10-03:
More importantly it should list  the options out in the usage or help, this is something the current make file does not do well.

cfbender, 2019-10-03:
@skullydazed what would you think of naming this command? Something like `qmk flash`? Though I could see that being used to only flash compiled firmware in the future. I may pick this one up once I finish #6884, if no one else is taking it

favorable-mutation, 2019-10-03:
Would be interested in picking up this or one of the other open CLI Issues, but not certain where to start; would someone mind linking me to instructions on setting up a development environment for CLI (linting, running unit tests, etc.)?

skullydazed, 2019-10-04:
@cfbender I think `qmk flash` is good. 

@favorable-mutation We have a page on [writing CLI commands](https://docs.qmk.fm/#/cli_development) and another one for our local conventions. If you have questions those don't cover feel free to raise them, either here or in a new issue. Documentation is always a work in progress. :)

favorable-mutation, 2019-10-04:
@skullydazed Awesome, tysm!

jorgemanzo, 2019-10-05:
@yanfali what would be a good way of listing out the make options that are available? Or is the list of options something that doesnt change often?

skullydazed, 2019-10-05:
@jorgemanzo The list does not change often. I don't think we have a full list that's easy to digest, but all of the methods to flash should be documented here: https://docs.qmk.fm/#/flashing

skullydazed, 2019-10-05:
The compiled firmware question is a good one. Since we are calling the make command we are implicitly compiling if needed, unless we add a check for the firmware file first. 

My default stance is to not add that check, but I have not really thought about it. Does anyone have an argument in favor of adding such a check?

jorgemanzo, 2019-10-05:
@skullydazed Im still new to this so correct me if this sounds odd. But if we compile, would it be easier to add a `--flash` cli argument to `compile` instead? I took a look at `compile` and it seems to do most of what is needed for this. The only thing missing is the `:<target>` for the `make` command?

skullydazed, 2019-10-05:
It's not odd at all, in fact it's one of the classic computer science struggles. When to write a separate program vs extending an existing program is not always easy to decide. In this case I want users to think of flashing as its own thing, so that eventually most people don't even bother with a separate compile step.

You're right that `qmk compile` does almost all of that already. Refactoring most of the code from `qmk.cli.compile` into appropriate modules under `qmk` is a good way to handle the duplication.

There are only a few ways to flash, they can simply be listed in the description for the argument.

jorgemanzo, 2019-10-05:
So I tried giving it a shot, and heres what I made: https://github.com/jorgemanzo/qmk_firmware/blob/4051bd4703dd9cb88847ed2cdc8c4fccc1386a2f/lib/python/qmk/cli/flash.py 

Something I still need to figure out is printing out the list when `qmk flash -h` is called. Currently in order to show the list, `qmk flash <bootloader>` needs to be called, which isnt really useful. If there was a way to check for a missing required argument, that would do it.

Anyway, heres how the subcommand is used:
`qmk flash [-h] [-kb KEYBOARD] [-km KEYMAP] [filename] bootloader`.

If this is progress, this would be my first contribution to an open source project, so feedback is appreciated!

skullydazed, 2019-10-06:
That is a good first step, but there is a lot of code duplication there between `flash` and `compile`.  It should be refactored so that less code is duplicated.

jorgemanzo, 2019-10-06:
I agree. When refactoring, should the common pieces of code between `qmk flash` and `qmk compile` be put in some common location to be used by both?

skullydazed, 2019-10-06:
Yes, we're slowly building up a library of useful python functions in `lib/python/qmk`. That's where the common functions should move to.

jorgemanzo, 2019-10-06:
Okay! I took another go at it and heres where Im at. I moved code that looked like was duplicated between `flash.py` and `compile.py` into `lib/python/qmk/commands.py`. https://github.com/jorgemanzo/qmk_firmware/commit/bcb009c488117247aef210c7e0c621227e7e2afa

Here is `compile.py` now using those methods: https://github.com/jorgemanzo/qmk_firmware/commit/4941700118eeb3d0b7e28083057de564df449c7a

rishav142k, 2019-10-07:
I am very much interested in CLI issues but not certain about very to begin with. Although I have tried various things,  I still can't. Any help would be appreciated.

stjohnjohnson, 2019-10-08:
What thoughts do people have on how to keep the options/args in sync between the CLI and the Makefile?

skullydazed, 2019-10-08:
@jorgemanzo I think you would benefit from some mentoring here. You are on the right track but there is still quite a bit to finish before that code is ready to submit in a PR. I am trying to provide guidance where I can but my time is limited right now.

You may be able to find some people with more time to mentor on our discord: https://discord.gg/Uq7gcHh

skullydazed, 2019-10-08:
@stjohnjohnson In the short term it will require some diligence. Longer term I think we can designate a single source of truth that everything else populates from.

stjohnjohnson, 2019-10-08:
For a first pass, you should just proxy all passed arguments directly to the makefile. Then you could easily iterate on it.

Shehanka, 2019-10-11:
This gonna be good issue can I take this

skullydazed, 2019-11-16:
Implemented in #6998

closed 2019-11-16 by skullydazed