qmk/qmk_firmware #6891: How to get keymaps to show up on the QMK configurator

blu006, 2019-10-03:
Hello,

I'm wondering if there is a way to get keymaps to show up by default on the QMK configurator.  Mainly, I want to set a default keymap for the handwired/tritium_numpad keyboard's ortho_6x4 layout.

Could this have something to do with the /layouts folder in the repo?

Thanks,
Brandon

yanfali, 2019-10-03:
@blu006 this is unfortunately a hard problem, parsing arbitrary C with macros is a non-trivial task. @skullydazed is working on a JSON driven version of QMK that will make this much easier in the future and a lot of these keymaps could then be converted and would be accessible from configurator. But at this time it's not very easy to do consistently.

blu006, 2019-10-03:
Is there a way to do it manually for now?  I'm more than willing to hand-translate my own keymaps until the automated JSON-driven QMK comes out.

yanfali, 2019-10-03:
kind of, you can simply use the configurator to copy the keymap you like and export a JSON file. The new format will be the same for the upcoming change.

blu006, 2019-10-03:
So saving the configuration and then uploading it to the /layouts folder in the repo would make it show up in the configurator eventually?

yanfali, 2019-10-03:
we don't currently support multiple keymaps per keyboard in the configurator at this time. You would need to store it locally. Eventually when everything is JSON driven it would be trivial to add that feature.

For now you'll just need to have a copy locally.

blu006, 2019-10-03:
Thanks for answering my questions.  I guess I'll close this PR until then
;)

danielo515, 2020-02-20:
If you use an ergodox I built a tool that allows you visualize your layout on a kind-of visual configurator: https://danielorodriguez.com/ergodox-configurator

closed 2022-08-04 by tzarc