qmk/qmk_firmware #6897: lib/lufa is empty downloading on Mac

ralfkaa, 2019-10-04:
I have been using qmk firmware for about a year now to experiment with musical layouts on keyboards.

had to reinstall my OS (High Sierra), and download the qmk_firmware again.

"make" in terminal no longer finds lufa content:

tmk_core/protocol/lufa.mk:14: lib/lufa/LUFA/makefile: No such file or directory
make[1]: *** No rule to make target `lib/lufa/LUFA/makefile'.  Stop.

in effect, lib/lufa is empty, whereas in a (Linux-based) backup there is quite a bit of content.

unfortunately not a programmer. am I missing something?

yanfali, 2019-10-04:
Hi @ralfkaa if you read our fine docs you will see you missed a step and need to type `make git-submodule`. Let us know if this doesn't resolve your issue

ralfkaa, 2019-10-04:
hi @yanfali, thanks for the quick response. 
I have been reading the docs regularly over the past year (docs.qmk.fm), yes, but don't understand your comment, nor how it relates to my issue. could you please send me a link to the specific spot in the docs, so that I can tell you whether or not it resolves my issue?
here's a bit more detail on what's happening:
on linux everything works as before, "make {keyboard}:{layout}". when I download the entire master files to my fedora machine, the lufa library is fully populated as with my previous Mac instance. problem is, I'm unable to successfully load the .hex file to my keyboard on fedora (as I said, I'm not a programmer, have to follow detailed instructions, and something is failing with installing the teensy loader on Fedora, possibly because the instructions at PJRC are for Ubuntu).
when I download the same master files to my high Sierra Mac machine, lib/lufa is empty, and "make {keyboard}:{layout}" delivers the error above. so I'm assuming something substantial has changed since I first downloaded the master files about a year ago, but only on Mac...
I hope I'm making sense.

yanfali, 2019-10-04:
@ralfkaa you deleted your entire env. So you need to follow the guide for setting up the repo for the very first time. https://docs.qmk.fm/#/newbs_getting_started

https://docs.qmk.fm/#/newbs_getting_started?id=set-up-qmk

Did you setup the submodules? since the last time you updated lufa was converted to a submodule. Hope this helps.

ralfkaa, 2019-10-05:
indeed, @yanfali , that did help!
since the remainder of the guide came after the linux instructions, I thought those steps were only applicable to linux, because last year all I had to do on the Mac was download the .zip and install home-brew, etc.
thank you for your assistance!

closed 2019-10-05 by ralfkaa

yanfali, 2019-10-06:
@ralfkaa awesome.

danielo515, 2020-01-25:
I am having this same problem if I move to master branch. It works on any other branch, and I didn't deleted my repo or changed it in any way other than pulling the latest master.
If I recall correctly the make submodules was an optional step. Is it now a mandatory one?

bestguy, 2020-05-24:
FYI if others see error on mac, you must git clone qmk, not download a zip.  `qmk setup` will run fine but your libs will be missing when you build:

```
$ qmk setup
Ψ Found qmk_firmware at ~/qmk_firmware.

*** Would you like to set ~/qmk_firmware as your QMK home? [y/n] y
...
Ψ Submodules are up to date.
Ψ QMK is ready to go

$ qmk compile -kb atomic -km default
Ψ Compiling keymap with make atomic:default

Making atomic with keymap default

tmk_core/protocol/lufa.mk:14: lib/lufa/LUFA/makefile: No such file or directory
make[1]: *** No rule to make target `lib/lufa/LUFA/makefile'.  Stop.
make: *** [atomic:default] Error 1
Make finished with errors
```

I didn't notice this docs so FYI

FlRENADO, 2020-11-18:
see i downloaded the .zip as just cloning it didnt work it would just sit there forever and do nothing at all no errors nothing so i downloaded the .zip and then ran qmk setup and everything went smoothly after that but i still get that exact error how am i suposed to fix it? im on windows (unfortunately) using msys2

yifleo, 2022-12-09:
I find the problem is files in qmk_firmware/lib/lufa are deleted ,after reset the deleted files in sourcetree app,I can complie successd.

MrNait95, 2023-01-08:
> I find the problem is files in qmk_firmware/lib/lufa are deleted ,after reset the deleted files in sourcetree app,I can complie successd.

how you did that? having the same problem here, quite noob on github btw

fauxpark, 2023-01-08:
Follow the newbs guide linked here: https://github.com/qmk/qmk_firmware/issues/6897#issuecomment-538541972