qmk/qmk_firmware #6905: Add support for 4 IS31FL3731 Drivers for RGB matrix

ZadenRB, 2019-10-05:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description
Currently, only 2 IS31FL3731 IC's are enabled for RGB matrix, while 4 are possible with LED matrix. Docs for RGB Matrix state `it would be trivial to support all 4 combinations`. I'd be interested in adding this support. It seems to me that all the `i2c_master` code is agnostic to the # of drivers, and the only edits needed would be adding defines for drivers 3&4 addresses and calling setup functions from `rgb_matrix_drivers` for those drivers. Am I missing anything?

drashna, 2019-10-12:
You'd also need to edit the actual driver file for it too, I think.

https://github.com/qmk/qmk_firmware/blob/23178d73fc94d28c10029171f205d4815e516270/drivers/issi/is31fl3731.c

ZadenRB, 2019-10-13:
@drashna I could be wrong, but it looks to me like all of the functions within the driver file take in either an address parameter, or loop based on `DRIVER_LED_TOTAL`, which is called from `rgb_matrix_drivers`, so I don't think that file actually needs to be edited.

drashna, 2019-10-17:
You're probably correct, and I noticed that too, after I posted.

tzarc, 2022-08-03:
Should be supported with `#define  DRIVER_COUNT 4` in config.h.

closed 2022-08-03 by tzarc