qmk/qmk_firmware #6911: CLI command to list keyboards

skullydazed, 2019-10-05:
## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

We should have a new command, `qmk list-keyboards`, that lists all the available keyboards. A good first step would be to simply run `make list-keyboards` and let it print to stdout.

It may be more performant to look at what make is doing and replicate the logic in python. If so we should do that so that `make list-keyboards` can call `qmk list-keyboards` instead of the current method.

danmcclain, 2019-10-05:
I'll take a quick pass at this today

closed 2019-10-07 by noroadsleft

raghavdhingra, 2019-10-23:
Looks good

PR #6927 (2019-10-07): Add `qmk list_keyboards`

## Description

`list_keyboards` replicates the `make list-keyboards` by globbing for all paths
that include `rules.mk` and then removing the paths that include `keymaps`.

This basis of this cli command could be reused in the future as a util, but is
not done so here since this would be the only place that would use it currently

Resolves #6911

## Types of Changes

- [x] Core
- [ ] Bugfix
- [x] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Clos [... cut]
commit f04e58da (2019-10-07): [CLI] Add `qmk list_keyboards` (#6927)

`list_keyboards` replicates the `make list-keyboards` by globbing for all paths
that include `rules.mk` and then removing the paths that include `keymaps`.

This basis of this cli command could be reused in the future as a util, but is
not done so here since this would be the only place that would use it currently

Resolves #6911