qmk/qmk_firmware #6912: CLI command to list keymaps

skullydazed, 2019-10-05:
## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

There is currently no way to list the available keymaps. We should have a `qmk list-keymaps` command that finds all the available keymaps for a given keyboard.

This one will take a bit of work to pull off. Besides finding all the keymaps in  `<keyboard_path>/keymaps` you'll have to check directories above that because the keyboard might be a sub-folder. Then you need to get the list of supported layouts from `rules.mk` and go trawling through the layouts to find those keymaps.

There is code to parse `rules.mk` here: https://github.com/qmk/qmk_compiler/blob/master/update_kb_redis.py#L172

And code to find all the keymaps (except for community layouts) here: https://github.com/qmk/qmk_compiler/blob/master/update_kb_redis.py#L401

My long term plan for the API involves moving most of that code into `qmk_firmware` anyway, so simply copying it into `lib/python/qmk` in a way that makes sense (with or without modifications) is a good path forward.

greenorb007, 2019-10-05:
@skullydazed I am ready to work on this please give me little direction(as i am little new to open source and all).

skullydazed, 2019-10-06:
@greenorb007 what sort of direction are you looking for?

pranshuchittora, 2019-10-06:
I would like to work on this. Any specificity before proceeding.
What exactly a list will contain.

alexweininger, 2019-10-06:
What do you mean by 
> you'll have to check directories above that because the keyboard might be a sub-folder
?

yanfali, 2019-10-06:
@alexweininger please take a look at the keyboard directory. Specifically: keebio, kbdfans, cannonkeys. These are vendor folders. Sometimes people have revisions per keyboard that share all the keymaps. Look at planck for an example. That means you won't find the keymaps in the keyboard directory, only the specific files for the revision of the hardware. Skully is trying to point out that to find the keymaps, you may need to keep searching recursively upwards until you hit a keymap directory.

Hope this helps.

alexweininger, 2019-10-06:
Yes this helps. I just wasn't sure what the case was for when a keymap file would be above the keyboard's directory. Thanks. @yanfali

danmcclain, 2019-10-08:
Is anyone actively working on this now? Was going to take a stab at it but don’t want to steal someone else’s thunder

stjohnjohnson, 2019-10-08:
@skullydazed it looks like you have a lot of folks interested in giving some contributions.  Would you prefer to assign it to someone or let N folks give competing PRs?

skullydazed, 2019-10-08:
We've historically had 2 problems with assigning these issues- If we don't assign them at all sometimes we get competing PR's. If we do assign them sometimes the volunteer doesn't follow through and then no one else works on the issue. To balance these competing factors we generally like to wait until there's both an open PR and a need to assign the issue.

Erovia, 2019-10-08:
Opened #6978 for this.

Shehanka, 2019-10-11:
Hope I solve this?

vaibhavbisht06, 2019-10-22:
Ready to work on this please give me brief

vaibhavbisht06, 2019-10-22:
Any specificity before proceeding

Erovia, 2019-10-22:
@vaibhavbisht06 PR #6978 already has a working implementation. If you have some ideas, check it out and advise.

PrathamGoyal, 2019-10-22:
I would like to work on this if it is not already assigned to someone. I have a basic idea about keymaps.

raghavdhingra, 2019-10-23:
Looks good

sakshamtaneja21, 2019-10-26:
i can help in this

closed 2022-08-03 by tzarc