qmk/qmk_firmware #6913: Windows subsystem for Linux in documentation

Gimly, 2019-10-05:
The documentation on Windows explains how to setup the build environment on Windows using MSYS2, which is nice but might be a bit complicated for new users. Wouldn't it make more sense to use the Windows subsystem for Linux (WSL) for this?

WSL is a real Linux that runs in Windows, so it should behave exactly like a Linux and has the advantage to be "built-in" the system since Windows 10.

alexweininger, 2019-10-06:
As a WSL user I would love to see this.

fauxpark, 2019-10-06:
I believe we do support WSL: https://docs.qmk.fm/#/getting_started_build_tools?id=toolchain-setup
This documentation was probably written at a time when WSL was not able to access USB devices for flashing, so MSYS2 was the recommended method instead.

noroadsleft, 2019-10-06:
Note that WSL is only available for Windows 10; Windows 8/8.1 users will need to use the MSYS2 method.

ThiagoWhispher, 2019-10-07:
It's a good idea, even with some details of Windows restrictions, it would still be another feature in favor of new users.

noroadsleft, 2019-10-07:
> It's a good idea, even with some details of Windows restrictions, it would still be another feature in favor of new users.

I agree. Just wanted it documented that the MSYS instructions need to remain if/when WSL instructions are added. :+1:

drashna, 2019-10-12:
Additionally, WSL does not allow you to flash the firmware for any of the bootloaders.   Additionally, last I checked, the 18.04 version of ubuntu for WSL still has broken packages for ARM GCC.   

Between these two issues, MSYS is a better solution, sadly.

sakshamtaneja21, 2019-10-26:
also can add this documentation to github and push over the website

drashna, 2019-12-01:
Closing this. 

Unfortunately, I don't think that WSL is in a state where it will be better than MSYS, in terms of usability. At least, not for everyone.

closed 2019-12-01 by drashna