qmk/qmk_firmware #6924: QMK Translations Hub

noroadsleft, 2019-10-05:
## QMK Firmware Documentation (docs.qmk.fm)

- Repository: [qmk/qmk_firmware](https://github.com/qmk/qmk_firmware/)
- Directory: https://github.com/qmk/qmk_firmware/tree/master/docs

### French

- Pull Requests:
  - [#6901](https://github.com/qmk/qmk_firmware/pull/6901) by @Gimly and @4sStylZ

### German

- Pull Requests:
  - [#6909](https://github.com/qmk/qmk_firmware/pull/6909) by @kuchosauronad0

----

## QMK Configurator (config.qmk.fm)

- Repository: [qmk/qmk_configurator](https://github.com/qmk/qmk_configurator/)
- Directory: https://github.com/qmk/qmk_configurator/tree/master/src/i18n

### Chinese

- Pull Requests: 
  - [#520](https://github.com/qmk/qmk_configurator/pull/520) by @mechmerlin

### French

- Pull Requests: 
  - **MERGED** [#325](https://github.com/qmk/qmk_configurator/pull/325) by @gab-a
- Additional Contributions by @zekth

### German

- Pull Requests:
  - [#513](https://github.com/qmk/qmk_configurator/pull/513) by @Vexatos
  - **MERGED** [#365](https://github.com/qmk/qmk_configurator/pull/365) by @heieisch (original translator)

### Russian

- Pull Requests:
  - [#519](https://github.com/qmk/qmk_configurator/pull/519) by @StrikerRUS and @nabokovas

### Spanish

- Pull Requests: 
  - **MERGED** [#516](https://github.com/qmk/qmk_configurator/pull/516) by @kbravh and @namelivia

----

## QMK Firmware website (qmk.fm)

- Repository: [qmk/qmk.fm](https://github.com/qmk/qmk.fm/)
- Directory: https://github.com/qmk/qmk.fm/tree/gh-pages/_i18n
- Instructions: https://github.com/qmk/qmk.fm/blob/gh-pages/readme.md

### German

- Pull Requests: 
  - [#60](https://github.com/qmk/qmk.fm/pull/60) by @Vexatos and @kuchosauronad0

### Russian

- Pull Requests: 
  - [#61](https://github.com/qmk/qmk.fm/pull/61) by @StrikerRUS and @nabokovas

kbravh, 2019-10-05:
I'm currently working on Spanish translations for the [qmk/qmk_firmware](https://github.com/qmk/qmk_firmware/) documentation.

Gimly, 2019-10-05:
@noroadsleft you can add the PR #6925 for the French translation, it adds the translation for the QMK Basics chapter

ThiagoWhispher, 2019-10-06:
@noroadsleft is it possible to add new languages ​​besides those mentioned above? I wanted to work on translating my native language (pt-BR)

kichloo, 2019-10-06:
I have made the Hindi translation of the same.

kichloo, 2019-10-06:
Here is the link of Hindi translation of CODE_OF_CONDUCT.md in Hindi
https://github.com/qmk/qmk_firmware/pull/6939

Kolkies, 2019-10-06:
I will look into translating the configurator!

ThiagoWhispher, 2019-10-06:
@noroadsleft I worked on the translation in the qmk/qmk_configurator repository for pt-BR available at https://github.com/qmk/qmk_configurator/pull/525

jackhumbert, 2019-10-06:
Sorry to the folks that commented - this was just meant to be a resource, not a discussion page. You can mention this issue in your PR (via `#6924`) and we'll get it added to the list :)

zvecr, 2022-08-28:
Closing as most translations have been removed due to being extremely stale, and this mostly covering hacktoberfest activities.

closed 2022-08-28 by zvecr