qmk/qmk_firmware #6944: Add the approx memory taken up by each feature in the docs

alexweininger, 2019-10-06:
This is a pure documentation request.

## Description

I think that it would be helpful on pages like the [Tap Dance Feature](https://docs.qmk.fm/#/feature_tap_dance) I think it might be useful to have an approximate for how much memory this feature might add to the firmware size in a more prominent location on each feature page.

For example on the tap dance page it is only mentioned in the doc. And others I wasn't able to find an approximate.

[image]

Thanks. Happy Hacktoberfest!

yanfali, 2019-10-06:
@thomasbaart care to contribute?

splitkb, 2019-10-06:
I made a guide on [how to save firmware space](https://thomasbaart.nl/2018/12/01/reducing-firmware-size-in-qmk/) at one point. Having that information be in the docs would be nice indeed, but is susceptible to enabling link time optimization and perhaps also by whether you'd use AVR or ARM.

I don't currently have the time to put into it, but there's at least a base to start from I suppose.

zvecr, 2019-10-06:
As a note to link things together, there is #6947 (and #5488) which does the opposite.

drashna, 2019-10-12:
I disagree with adding the firmware size for each feature.  At best, we should put a ballpark size. 

The reasoning behind this: 
* Different OS' use different versions of GCC, and different version compile different.  8.3 compiles to the smallest size, while 9.x is VERY large by comparison, even more so than 5.x or 6.x
* Some features have specific exceptions/handling based on what other features are enabled. Also, some code may be optimized away (due to redundancy) when multiple features are enabled.  And vice versa, too. 
* Link Time Optimization drastically reduces the compile size in many cases, dropping the size considerably. 

Because of these issues, having a stated size doesn't make sense, IMO, because then we have moving targets that may or may not be correct.... which is precisely why we have PRs removing these numbers.

sakshamtaneja21, 2019-10-26:
i can help in this

tzarc, 2022-08-03:
Going to go ahead and close this; different MCUs, different compilers, different compilation flags -- far too many combinations to keep track of, and far too many to keep up to date.

Ultimately, people only find a problem when things don't fit -- they very rarely if ever "plan to fit features X, Y, and Z" based on arbitrary firmware sizes.

closed 2022-08-03 by tzarc