qmk/qmk_firmware #6953: TAPPING_TERM counting the second key hold time is unintuitive

xster, 2019-10-07:
TAPPING_TERM and variations like PERMISSIVE_HOLD IGNORE_MOD_TAP_INTERRUPT seem to facilitate things for making more things register as holds, but I'd argue the default TAPPING_TERM is already over-inclusive in one case for "slow" typists:

Assume a TAPPING_TERM of 200ms

t=0ms - press down KC_RCTL
t=20ms - press down KC_A
t=40ms - KC_RTCL up
t=220ms - KC_A up

Even though the mod key KC_RCTL was held down very briefly, it still gets registered as a hold since the second key was held down for longer. 

For slow typists who'd like to keep the TAPPING_TERM at the threshold of their natural typing speed, this doubles the odds of a false positive and I'm also not sure what the general valid use case this behavior is meant to support.

xster, 2019-10-07:
Actually playing around with this a bit more, it seems like this is explicitly meant to be fixed by IGNORE_MOD_TAP_INTERRUPT but I would have never realized it from reading the doc. 

I didn't read through the source and I'm only testing this empirically but it seems like the IGNORE_MOD_TAP_INTERRUPT behavior is almost the reverse of how the doc describes it. i.e. defining it makes hold triggers less likely rather than more likely. 

Tentatively sending https://github.com/qmk/qmk_firmware/pull/6955 to verify my understanding.

xster, 2019-10-07:
Looking through the codebase, it seems like there's a lot of other duplicate 'solutions' to this same problem in their own config.h https://github.com/qmk/qmk_firmware/blob/a1a7a84831230515bac68c9fe2c93cd90c02acb8/keyboards/georgi/config.h#L59 https://github.com/qmk/qmk_firmware/blob/4be1dfcee48daeb77e966b08ad9ca51e02253acf/users/manna-harbour_miryoku/config.h#L6 https://github.com/qmk/qmk_firmware/blob/cc8744548108b1c1f820fb411ac3bf313aca1f5f/keyboards/jd45/keymaps/blakedietz/config.h#L5

where they also felt the need to 'redocument' what this flag does. 

Arguably the behavior with `IGNORE_MOD_TAP_INTERRUPT` should be the default behavior. As mentioned above, I can't imagine a valid proper use case for it. But I can imagine how big of a breaking change it would be changing the default.

rpbaptist, 2019-10-07:
I just want to weigh in on this, because I also think the description is very confusing. Each time I had to revisit these options I had to re-read the docs and only the provided examples allowed me to understand it. It took a lot of testing with `TAPPING_TERM`, `PERMISSIVE_HOLD` and `IGNORE_MOD_TAP_INTERRUPT`to get a situation where the keyboard behaved to my liking.

In fact, I was convinced that I needed `PERMISSIVE_HOLD` and not `IGNORE_MOD_TAP_INTERRUPT` to solve the issues I had until trial and error proved otherwise.

xster, 2019-10-08:
Another potential piece of supporting data, someone at ergodox-ez probably thought the default behavior (IGNORE_MOD_TAP_INTERRUPT undefined) is confusing and flipped it. 

[image]

noroadsleft, 2019-10-08:
@drashna Could you weigh in on this?

xster, 2020-01-08:
Would like to get #6955 reviewed if anyone has the time. I think it adds more clarification.

closed 2020-05-07 by stale[bot]