qmk/qmk_firmware #6957: QMK FAQ contains empty section about the "Quantum" in the name

Maxr1998, 2019-10-07:
In the [QMK FAQ](https://github.com/qmk/qmk_firmware/blob/master/docs/faq_general.md), there is a section "Why the Name Quantum?". Instead of an answer, there is only a `` in there. I think that that section should either be expanded with an answer, or removed. Additionally, it should only have two `#` instead of three, to bring it into one level with the other items.

jackhumbert, 2019-10-07:
I think I'm going to blame @skullydazed's [commit](https://github.com/qmk/qmk_firmware/commit/5810bb67e4280582ef2811a7246f90d5f0bf292d) here :) I think it could probably be removed, unless we want to explain the Quantum Mechanics => Mechanical Keyboard play on words. It'll really just be a bit of trivia going forward.

Maxr1998, 2019-10-07:
That explains a lot ^^
I also think it could be removed, it's not that relevant to using the firmware, and while it might be interesting to some, it's not a "frequently asked question". I will file a PR in a minute.

commit 5bb3fe7a (2019-10-07): Remove unanswered/unnecessary FAQ item

As discussed in #6957, closes #6957
closed 2019-10-07 by skullydazed