qmk/qmk_firmware #6974: Can't build if mousekeys are in constant mode

StephaneLeveugle, 2019-10-08:
## Describe the Bug

Attempting to build the firmware with mousekeys in constant mode gives me this error:
``` $ make preonic/rev3:stephaneleveugle
QMK Firmware 0.7.27
Making preonic/rev3 with keymap stephaneleveugle

arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   49056       0   49056    bfa0 .build/preonic_rev3_stephaneleveugle.hex

Compiling: quantum/audio/muse.c                                                                     [OK]
Compiling: keyboards/preonic/preonic.c                                                              [OK]
Compiling: keyboards/preonic/rev3/rev3.c                                                            [OK]
Compiling: keyboards/preonic/keymaps/stephaneleveugle/keymap.c                                      [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_g.c                                                                 [OK]
Compiling: quantum/process_keycode/process_audio.c                                                  [OK]
Compiling: quantum/process_keycode/process_clicky.c                                                 [OK]
Compiling: quantum/audio/audio_arm.c                                                                [OK]
Compiling: quantum/audio/voices.c                                                                   [OK]
Compiling: quantum/audio/luts.c                                                                     [OK]
Compiling: quantum/process_keycode/process_music.c                                                  [OK]
Compiling: quantum/encoder.c                                                                        [OK]
Compiling: quantum/process_keycode/process_space_cadet.c                                            [OK]
Compiling: quantum/dip_switch.c                                                                     [OK]
Compiling: tmk_core/common/host.c                                                                   [OK]
Compiling: tmk_core/common/keyboard.c                                                               [OK]
Compiling: tmk_core/common/action.c                                                                 [OK]
Compiling: tmk_core/common/action_tapping.c                                                         [OK]
Compiling: tmk_core/common/action_macro.c                                                           [OK]
Compiling: tmk_core/common/action_layer.c                                                           [OK]
Compiling: tmk_core/common/action_util.c                                                            [OK]
Compiling: tmk_core/common/print.c                                                                  [OK]
Compiling: tmk_core/common/debug.c                                                                  [OK]
Compiling: tmk_core/common/util.c                                                                   [OK]
Compiling: tmk_core/common/eeconfig.c                                                               [OK]
Compiling: tmk_core/common/report.c                                                                 [OK]
Compiling: tmk_core/common/chibios/suspend.c                                                        [OK]
Compiling: tmk_core/common/chibios/timer.c                                                          [OK]
Compiling: tmk_core/common/chibios/bootloader.c                                                     [OK]
Compiling: tmk_core/common/chibios/printf.c                                                         [OK]
Compiling: tmk_core/common/chibios/eeprom_stm32.c                                                   [OK]
Compiling: tmk_core/common/chibios/flash_stm32.c                                                    [OK]
Compiling: tmk_core/common/bootmagic.c                                                              [OK]
Compiling: tmk_core/common/mousekey.c                                                               [OK]
Compiling: tmk_core/common/command.c                                                               tmk_core/common/command.c: In function 'mousekey_console_help':
tmk_core/common/command.c:721:13: error: 'MOUSEKEY_MOVE_DELTA' undeclared (first use in this function)
             MOUSEKEY_MOVE_DELTA, MOUSEKEY_WHEEL_DELTA);
             ^~~~~~~~~~~~~~~~~~~
tmk_core/common/command.c:721:13: note: each undeclared identifier is reported only once for each function it appears in
tmk_core/common/command.c:721:34: error: 'MOUSEKEY_WHEEL_DELTA' undeclared (first use in this function)
             MOUSEKEY_MOVE_DELTA, MOUSEKEY_WHEEL_DELTA);
                                  ^~~~~~~~~~~~~~~~~~~~
tmk_core/common/command.c: In function 'mousekey_console':
tmk_core/common/command.c:764:36: error: 'MOUSEKEY_DELAY' undeclared (first use in this function)
             mk_delay             = MOUSEKEY_DELAY / 10;
                                    ^~~~~~~~~~~~~~
tmk_core/common/command.c:765:36: error: 'MOUSEKEY_INTERVAL' undeclared (first use in this function)
             mk_interval          = MOUSEKEY_INTERVAL;
                                    ^~~~~~~~~~~~~~~~~
tmk_core/common/command.c:766:36: error: 'MOUSEKEY_MAX_SPEED' undeclared (first use in this function)
[... 122 lines cut ...]
 | mypath/qmk_firmware/tmk_core/common/command.c:649: undefined reference to `mk_interval'
 | mypath/qmk_firmware/tmk_core/common/command.c:649: undefined reference to `mk_max_speed'
 | mypath/qmk_firmware/tmk_core/common/command.c:649: undefined reference to `mk_time_to_max'
 | mypath/qmk_firmware/tmk_core/common/command.c:649: undefined reference to `mk_wheel_max_speed'
 | mypath/qmk_firmware/tmk_core/common/command.c:649: undefined reference to `mk_wheel_time_to_max'
 | .build/obj_preonic_rev3_stephaneleveugle/common/command.o: In function `mousekey_param_dec':
 | mypath/qmk_firmware/tmk_core/common/command.c:696: undefined reference to `mk_delay'
 | mypath/qmk_firmware/tmk_core/common/command.c:696: undefined reference to `mk_interval'
 | mypath/qmk_firmware/tmk_core/common/command.c:696: undefined reference to `mk_max_speed'
 | mypath/qmk_firmware/tmk_core/common/command.c:696: undefined reference to `mk_time_to_max'
 | mypath/qmk_firmware/tmk_core/common/command.c:696: undefined reference to `mk_wheel_max_speed'
 | mypath/qmk_firmware/tmk_core/common/command.c:696: undefined reference to `mk_wheel_time_to_max'
 | .build/obj_preonic_rev3_stephaneleveugle/common/command.o: In function `mousekey_console':
 | mypath/qmk_firmware/tmk_core/common/command.c:779: undefined reference to `mk_delay'
 | mypath/qmk_firmware/tmk_core/common/command.c:779: undefined reference to `mk_interval'
 | mypath/qmk_firmware/tmk_core/common/command.c:779: undefined reference to `mk_max_speed'
 | mypath/qmk_firmware/tmk_core/common/command.c:779: undefined reference to `mk_time_to_max'
 | mypath/qmk_firmware/tmk_core/common/command.c:779: undefined reference to `mk_wheel_max_speed'
 | mypath/qmk_firmware/tmk_core/common/command.c:779: undefined reference to `mk_wheel_time_to_max'
 | collect2.exe: error: ld returned 1 exit status
 |
make[1]: *** [tmk_core/rules.mk:300: .build/preonic_rev3_stephaneleveugle.elf] Error 1
Make finished with errors
make: *** [Makefile:574: preonic/rev3:stephaneleveugle] Error 1
```

Kikangh, 2019-10-10:
Hi,

Might be a stupid question on my side, but did you enable the mousekey in your rules.mk ?

MOUSEKEY_ENABLE = yes

StephaneLeveugle, 2019-10-10:
Hi,

Yes I did!
Sorry I forgot to mention it :)

drashna, 2019-10-12:
Well, the simple solution right now would be to disable the console, as that appears to be what is causing the compiler issue. 

However, it looks like there is definitely a bug here, right now.

drashna, 2019-10-12:
I have a PR open that should fix this issue.  You can check out #7017, if you want

PR #7017 (2020-05-19): Disable Mousekey Command interface when Constant speed is enabled
This is kind of a brute force fix, but in "Constant Speed" mode, these values don't really change, for the ones that still exist.  

This disables the mousekey console/command code if this mode is deactivated, fixing the compiler errors that are generated. 

# Types of Changes

- [x] Core
- [x] Bugfix

## Issues Fixed or Closed by This PR

* Fixes #6974

## Checklist
- [x] My code follows the code style of this project.
- [x] I have read the [**CONTRIBUTING** document](https://docs.qmk.fm/#/contributing).
- [x] I have tested the [... cut]
closed 2020-05-19 by nooges