qmk/qmk_firmware #6976: Problems compiling for a Ergodox Planck_EZ

michaeleherman, 2019-10-08:
I'm trying to compile my keymap for my Planck\_Ez. It used to work but isn't now. The error looks like this:

```
QMK Firmware 0.7.34
Making planck/ez with keymap NotVeryCleverOne

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: quantum/audio/muse.c                                                                    In file included from ./lib/chibios/os/rt/include/ch.h:91:0,
                 from ./lib/chibios/os/hal/osal/rt/osal.h:32,
                 from ./lib/chibios/os/hal/include/hal.h:28,
                 from quantum/quantum.h:24,
                 from quantum/audio/muse.h:4,
                 from quantum/audio/muse.c:1:
./lib/chibios/os/rt/include/chchecks.h:42:2: error: #error "obsolete or unknown configuration file"
 #error "obsolete or unknown configuration file"
  ^~~~~
./lib/chibios/os/rt/include/chchecks.h:55:2: error: #error "CH_CFG_INTERVALS_SIZE not defined in chconf.h"
 #error "CH_CFG_INTERVALS_SIZE not defined in chconf.h"
  ^~~~~
./lib/chibios/os/rt/include/chchecks.h:59:2: error: #error "CH_CFG_TIME_TYPES_SIZE not defined in chconf.h"
 #error "CH_CFG_TIME_TYPES_SIZE not defined in chconf.h"
  ^~~~~
./lib/chibios/os/rt/include/chchecks.h:158:2: error: #error "CH_CFG_USE_OBJ_FIFOS not defined in chconf.h"
 #error "CH_CFG_USE_OBJ_FIFOS not defined in chconf.h"
  ^~~~~
./lib/chibios/os/rt/include/chchecks.h:167:2: error: #error "CH_CFG_USE_FACTORY not defined in chconf.h"
 #error "CH_CFG_USE_FACTORY not defined in chconf.h"
  ^~~~~
./lib/chibios/os/rt/include/chchecks.h:171:2: error: #error "CH_CFG_FACTORY_MAX_NAMES_LENGTH not defined in chconf.h"
 #error "CH_CFG_FACTORY_MAX_NAMES_LENGTH not defined in chconf.h"
  ^~~~~
./lib/chibios/os/rt/include/chchecks.h:175:2: error: #error "CH_CFG_FACTORY_OBJECTS_REGISTRY not defined in chconf.h"
 #error "CH_CFG_FACTORY_OBJECTS_REGISTRY not defined in chconf.h"
  ^~~~~
./lib/chibios/os/rt/include/chchecks.h:179:2: error: #error "CH_CFG_FACTORY_GENERIC_BUFFERS not defined in chconf.h"
 #error "CH_CFG_FACTORY_GENERIC_BUFFERS not defined in chconf.h"
  ^~~~~
./lib/chibios/os/rt/include/chchecks.h:183:2: error: #error "CH_CFG_FACTORY_SEMAPHORES not defined in chconf.h"
 #error "CH_CFG_FACTORY_SEMAPHORES not defined in chconf.h"
  ^~~~~
./lib/chibios/os/rt/include/chchecks.h:187:2: error: #error "CH_CFG_FACTORY_MAILBOXES not defined in chconf.h"
 #error "CH_CFG_FACTORY_MAILBOXES not defined in chconf.h"
  ^~~~~
./lib/chibios/os/rt/include/chchecks.h:191:2: error: #error "CH_CFG_FACTORY_OBJ_FIFOS not defined in chconf.h"
 #error "CH_CFG_FACTORY_OBJ_FIFOS not defined in chconf.h"
  ^~~~~
./lib/chibios/os/rt/include/chchecks.h:233:2: error: #error "CH_CFG_SYSTEM_INIT_HOOK not defined in chconf.h"
 #error "CH_CFG_SYSTEM_INIT_HOOK not defined in chconf.h"
  ^~~~~
./lib/chibios/os/rt/include/chchecks.h:237:2: error: #error "CH_CFG_SYSTEM_EXTRA_FIELDS not defined in chconf.h"
 #error "CH_CFG_SYSTEM_EXTRA_FIELDS not defined in chconf.h"
  ^~~~~
In file included from ./lib/chibios/os/rt/include/ch.h:114:0,
                 from ./lib/chibios/os/hal/osal/rt/osal.h:32,
                 from ./lib/chibios/os/hal/include/hal.h:28,
                 from quantum/quantum.h:24,
                 from quantum/audio/muse.h:4,
                 from quantum/audio/muse.c:1:
./lib/chibios/os/rt/include/chschd.h:429:3: error: expected specifier-qualifier-list before 'CH_CFG_SYSTEM_EXTRA_FIELDS'
   CH_CFG_SYSTEM_EXTRA_FIELDS
   ^~~~~~~~~~~~~~~~~~~~~~~~~~
 [ERRORS]
 |
 |
 |
make[1]: *** [.build/obj_planck_ez_NotVeryCleverOne/muse.o] Error 1
make: *** [planck/ez:NotVeryCleverOne] Error 1
```

I looked at the current open issues but didn't find anything. I also tried a make clean and make git-submodule.

Any help will be greatly appreciated.

laurentlaurent, 2019-10-09:
Seems to be a version mismatch between your Chibios and the chconf.h: http://www.chibios.com/forum/viewtopic.php?t=5098

I also saw that the version of chibios folder you have on your qmk fork is different from the one in qmk master

drashna, 2019-10-12:
Yeah, it looks like you changed the submodules.  Doing so can break things,as it has here.  You'd need to revert the changes.

michaeleherman, 2019-10-12:
Thanks. I simply cloned the qmk repo again and my compile worked.

closed 2019-10-12 by michaeleherman