qmk/qmk_firmware #6991: ` key sets to the Esc for DZ60

JTP709, 2019-10-09:
I've been trying to set a key to the backtick ` / tilde ~ but after flashing it to my DZ60 board from KBDFans.cn it functions as the Escape key, regardless of what key I try to map it to. I have the DZ60 with 2 shift keys and split backspace key. I'm trying to set the split backspace key the backtick.

drashna, 2019-10-12:
@JTP709 did you have a link for your keymap?

yanfali, 2019-10-14:
@JTP709 clear your eeprom, there's a bootmagic setting that flips tilde and esc. It's probably just some eeprom corruption.

daskygit, 2021-09-28:
> I'm having the same issue with my DZ60RGB (new boot loader v2.1), `KC_ESC` works as expected, `KC_GESC` is simply ignored i.e., no keystroke registered. I've tried to clear the eeprom with `EEP_RST` but without success.
> 
> Note that I have this problem only with this board, all other boards are working as designed with `KC_GESC`

In this case it's disabled in the rules for the dz60rgb v2_1 https://github.com/qmk/qmk_firmware/blob/02451fcc7507128ea7bded71962e19df2b2ecb02/keyboards/dztech/dz60rgb/v2_1/rules.mk#L28-L29
You'll need to compile locally and enable it in your rules. You'll likely need to disable something else as the new bootloader limits the available size.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc

34ounce, 2023-06-03:
I joined git because of this.
try to use dz64rgb layout to solve esc/tilde key problem
https://config.qmk.fm/#/dztech/dz64rgb/LAYOUT_64_ansi 
Here is example that I just uploaded 
https://github.com/34ounce/dz60rgb/commit/d551f1883c4ea64cf271fa7c1bfa28f2f0d31968
Hope it helps someone with dz64rgb.

34ounce, 2023-06-04:
I joined git because of this.
try to use preset dz64rgb ansi layout instead of dz64_v2_1
dz64rgb qmk conf link: 
https://config.qmk.fm/#/dztech/dz64rgb/LAYOUT_64_ansi 
Here is example:
https://github.com/34ounce/dz60rgb/blob/main/dz60rgb_v2_1.json
Hope it helps someone with dz64rgb.