qmk/qmk_firmware #6992: Unable to clone forked repo

mrpbennett, 2019-10-09:
## Describe the Bug

Unable to clone the repo once forked via SSH or HTTPS.

➜ git clone https://github.com/mrpbennett/qmk_firmware.git
Cloning into 'qmk_firmware'...
remote: Enumerating objects: 182898, done.
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system:
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

noroadsleft, 2019-10-09:
I've never seen or experienced this before, but the tips on this page look promising:

- [The remote end hung up unexpectedly while git cloning - Stack Overflow](https://stackoverflow.com/questions/6842687/the-remote-end-hung-up-unexpectedly-while-git-cloning)

mrpbennett, 2019-10-09:
> I've never seen or experienced this before, but the tips on this page look promising:
> 
> * [The remote end hung up unexpectedly while git cloning - Stack Overflow](https://stackoverflow.com/questions/6842687/the-remote-end-hung-up-unexpectedly-while-git-cloning)

Thanks for that...is that just a cmd line entry...or does this go into my git config file?

yanfali, 2019-10-09:
@mrpbennett this is probably something local to you, like you're running out of resources. You could also check if github is have an outage, but it's not really a QMK issue as such.

noroadsleft, 2019-10-09:
@mrpbennett,

> Thanks for that...is that just a cmd line entry...or does this go into my git config file?

Technically, it's both. Just enter `git config --global http.postBuffer 524288000` into your CLI and try cloning again. Using that command will actually write the setting to your .gitconfig file and use it going forward.

mrpbennett, 2019-10-09:
I just checked my .gitconfig my buffer size is larger already

[http]
	postBuffer = 1048576000

saesh, 2019-10-10:
This looks similar to #6784. Are you using MinGW as well?

closed 2020-02-07 by stale[bot]

kenhoulin, 2020-05-23:
running into the same issue here...