qmk/qmk_firmware #7003: Enabling NKRO breaks all modifiers - space, enter, control, etc.

issmirnov, 2019-10-10:
## Describe the Bug

Team, I am completely stumped. Whenever I enable NKRO (using the appropriate `#define FORCE_NKRO` and `NKRO_ENABLE = yes`) my modifier keys stop responding. 

This has been an issue for about 6 months. I seem to run into this mostly on my mac, but it's been happening on my linux box lately as well. I've tried clearing EEPROM, reflashing the firmware, etc. 

What's most interesting is this happens on 2 different keyboards, and persists across avr-gcc versions and host OS's.

## System Information

 - Keyboard: 
    - Ergodox EZ (Classic) 
    - custom Keebio Levinson Let's Split
 - Operating system: 
    - Mac OS 10.13, MacOS 10.14
    - Debian 10, ubuntu 18.04.01
 - AVR GCC version: 
     - MacOS: 9.2.0
     - Linux: 5.4.0
 - ARM GCC version: 
    - MacOS: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
    - Linux: arm-none-eabi-gcc (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
 - QMK Firmware version: `0.7.36`. See also https://github.com/issmirnov/qmk-keebs for version lock.
 - Any keyboard related software installed? 
   - [X] Karabiner. However, have tried without this and same issue.

## Additional Context

My repo is located at https://github.com/issmirnov/qmk-keebs. Sample build:

```
git clone https://github.com/issmirnov/qmk-keebs ismirnov-keebs
cd ismirnov-keebs
git submodule init
cd qmk_firmware && make git-submodule

git revert 44c95df # enable NKRO again.

# ergodox
./qmk build ergodox_ez
./qmk flash ergodox_ez

# levinson
./qmk build levinson
./qmk flash levinson
```

Any help is appreciated. Let me know if you have any questions about my repo layout - should be mostly self-explanatory.

Thanks all!

drashna, 2019-10-12:
That's very odd, and I can't reproduce, actually. 

For the avr-gcc version, 9.2 has issues and that may be part of the problem.

issmirnov, 2019-10-13:
Hmm, thanks for looking into this. I get the same result on a linux install running v5.4.0. 

Do you want me to upload my hex build and see if there's a diff there? Were you able to build using my instructions?

yanfali, 2019-10-14:
Question, you seem to have a very customized keymap. What happens if you use default. Does the problem still happen?

issmirnov, 2019-10-15:
@yanfali the result is the same. You can see my exact code at the [debug](https://github.com/issmirnov/qmk-keebs/tree/debug) branch. Using the default keymap from ergodox EZ (see https://github.com/issmirnov/qmk-keebs/commit/4c057381322c7ae35824fde3d80e7f7411a0e604).

issmirnov, 2019-10-15:
Just for run, I checked that removing the `LINK_TIME_OPTIMIZATION_ENABLE = yes` rule has no effect. I made sure to run `make clean` in the qmk_firmware root.

Any thoughts folks? I can upload my binaries and we can diff them to try to figure out why reproducing is hard. Symptoms remain constant: space, backspace, enter, and tab break. I haven't tested other modifiers.

jes5199, 2020-04-26:
I'm seeing this bug too, with a planck/rev6 using the default layout from QMK Toolbox, on a mac

robertofarrell, 2020-05-08:
I'm seeing this as well with all my `projectkb` boards.  Disabling NKRO seems to resolve it though.

jes5199, 2020-05-08:
(I should have mentioned that my keyboard works fine if NKRO is disabled)

jvbroek, 2021-03-09:
Adding my $0.02 here - same issue, board works perfectly with NKRO set to off in rules.mk, only alphanumerics (and some modifiers) work without it.  Running STM32L072, compiled on a mac.

closed 2022-08-29 by zvecr