qmk/qmk_firmware #7033: "make" failure on Mac: "magic_ee_hands" error in quantum.c

ralfkaa, 2019-10-14:
## Describe the Bug

any attempt to make keyboard:layout aborts with the following error message:

[begin pasted terminal output]

Compiling: quantum/quantum.c                                                                       quantum/quantum.c: In function 'process_record_quantum':
quantum/quantum.c:543:39: error: 'MAGIC_EE_HANDS_RIGHT' undeclared (first use in this function)
         case MAGIC_SWAP_LCTL_LGUI ... MAGIC_EE_HANDS_RIGHT:
                                       ^~~~~~~~~~~~~~~~~~~~
quantum/quantum.c:543:39: note: each undeclared identifier is reported only once for each function it appears in
quantum/quantum.c:661:26: error: 'MAGIC_EE_HANDS_LEFT' undeclared (first use in this function)
                     case MAGIC_EE_HANDS_LEFT:
                          ^~~~~~~~~~~~~~~~~~~
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [.build/obj_xd75_default/quantum/quantum.o] Error 1
make: *** [xd75:default] Error 1
Make finished with errors

[end terminal pasted output]

## System Information
MacBook Air 4,1
 - Keyboard: any keyboard  
 - Operating system: MacOS High Sierra
 - AVR GCC version: 8.3.0

 - ARM GCC version: (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 - QMK Firmware version: 0.7.32

 - Any keyboard related software installed? No
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

yanfali, 2019-10-14:
Hi, you need to publish a repo that has the source code. Right now this looks like a programming error on your side and isn't something we can help you with. Please jump on the discord for that kind of help. Issues are really for problems with the actual QMK code, not problems with your custom versions. Thanks

ralfkaa, 2019-10-14:
thank you, Yan Fa Li 
not sure what you mean by "custom versions". I've tested the "make" with several preexisting keyboards and layouts, and the error occurs with all of them. there's something faulty with the actual git clone.
I'll do some reinstalling to see whether the issue disappears.
fyi, I wrote recently about the total failure of "make" on Mac, and you pointed the way to cloning the git with the libraries. at first "make" was working (last week). suddenly it's not... I don't understand enough about a git clone (or programming, for that matter) to figure out how that is possible, so I will have to do some further research.
thank you.

closed 2019-10-14 by ralfkaa