qmk/qmk_firmware #7040: "pip install" failed in script install_linux.sh

netcrop, 2019-10-15:
"pip install" failed in script install_linux.sh

## Describe the Bug

1, "pip install" failed in script install_linux.sh due to python-pip isn't a package dependency for archlinux.
This can be fixed by add python-pip as a package dependency for archlinux.
2, "pip install" shows ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/usr/lib/python3.7/site-packages/argcomplete'
This can be fixed by using "sudo pip install".
## System Information
Tested on Linux arch 5.3.5-arch1-1-ARCH
## Additional Context

rishav142k, 2019-10-15:
Try setting up a virtual environment and then installing the package in it.

netcrop, 2019-10-16:
It turns out all the python requirements already are included in archlinux official repositories. "pacman -S --needed python-appdirs python-argcomplete python-colorama" will solve the issue for archlinux.

closed 2019-10-19 by netcrop