qmk/qmk_firmware #7047: Port PloopyCo Trackball into QMK Firmware

drashna, 2019-10-16:
For anyone that hasn't seen it yet, /u/crop_octagon build a DIY Trackball. 
https://www.reddit.com/r/MechanicalKeyboards/comments/di8va1/opensource_trackball_project_completed_source/

Arduino code here:
https://github.com/ploopyco/mouse

This uses an ATmega32u4, 8MHz 3.3v chip.  Meaning that we can flash QMK Firmware on it. 

The buttons use a direct pin type approach, and the scroll wheel seems to be compatible with the rotary encoder feature, but something more custom may be better then. 

And it uses a PMW3360 optical sensor, hooked up with SPI. 

I have this started, but if anyone wants to help/contribute by all means!
https://github.com/drashna/qmk_firmware/tree/trackball/ploopyco/keyboards/mouse/ploopyco_trackball

gyvess, 2019-10-20:
Would it be possible to include advanced mouse acceleration options in QMK?

Reasons why you'd might want to do this:

- Acceleration independent of not just game performance but entire system performance
- Anti-cheat false positive detection approaching zero, as no system software has to be altered
- Portability. Have your trackball movement behave exactly as you want it anywhere, assuming you disable the appropriate userland mouse settings

As for what settings would be necessary or how acceleration works, KovaaK's explains things better than I could. He has created software for acceleration as well as made blog posts and videos explaining it so I'll just link those now.

[Introduction to Mouse Accel for competitive gaming](https://mouseaccel.blogspot.com/2015/11/introduction-to-mouse-accel-for.html)
[Interaccel - Program for enabling advanced mouse acceleration](https://github.com/KovaaK/InterAccel)
[Video explaining acceleration settings and showing how they function in practice](https://www.youtube.com/watch?v=PmY1OTacEzA)

drashna, 2019-10-26:
@gyvess yes and no.  I'm sure that some this is possible, but that's not the goal here.  It's to get it working, not fine tuning it.  That can come later.

And we're not using any sort of processing right now, other than to make sure that it's angled correctly.  

Aside from that, we're just using this: https://docs.qmk.fm/#/feature_pointing_device

drashna, 2020-10-06:
This is completed, and with the PloopyCo mouse, as well.  

Closing.

closed 2020-10-06 by drashna