qmk/qmk_firmware #7049: Macro to output Windows Alt Codes

ransworld, 2019-10-16:
Hello,

I'm hoping to output some special characters with a single keystroke on my Planck EZ, such as _em dash_ (—) and _en dash_ (–), so they can be typed easily in Windows. I'm not sure what the best method would be, but I've been trying to create a macro that replicates the Alt code input. Basically, I need to enter a sequence of _numpad_ keys while holding the _alt_ key.

Following the [QMK Macros](https://beta.docs.qmk.fm/features/feature_macros) guide, I have been able to enter a string of text while the alt key is down, with:

```
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
  switch (keycode) {
    case EMDASH:
      if (record->event.pressed) {
        SEND_STRING(SS_LALT("0151"));
      } else {
      }
      break;
  }
  return true;
}
```

Instead of the string of numbers (0151), I need to enter the numpad keycodes. How do I format this?

drashna, 2019-10-17:
You're looking for unicode:
https://docs.qmk.fm/#/feature_unicode

And for macros: https://docs.qmk.fm/#/feature_unicode?id=send_unicode_hex_string

ransworld, 2019-10-17:
Unicode was looking promising; but it looks like I would need to install WinCompose on the terminal server at work, which isn't an option.

I've figured it out though:
`SEND_STRING(SS_LALT(SS_TAP(X_KP_0) SS_TAP(X_KP_1) SS_TAP(X_KP_5) SS_TAP(X_KP_1)));`

I should have been using the `X_` prefix, not the `KC_` prefix.

closed 2019-10-17 by ransworld