qmk/qmk_firmware #7065: Characters are being typed randomly

nschurmann, 2019-10-17:
Yesterday I assembled a Corne keyboard. It has two pro micros in it as it's a split keyboard. The left side... when connected by itself works ok. However the right side of the keyboard it's acting weird. I will try to describe it as follows:

When you first plug in the keyboard it works fine... it gets all your keystrokes and you can use it with ease. This keyboard also has some oled panels.
However... randomly, the keyboard will start typing caracters for you. This is very random so i think that, if the assembly was somewhere bridged, then it should be constant... however the keyboard it's usable for a couple of minutes, sometimes only seconds. The keys that are usually randomly pressed are this as in a qwerty keyboard:
Y
H
N

As you can see is the first column of the keyboard... or the 7th column of a complete keyboard... i checked the solder but it seems ok to me.

Have anyone have had a similar issue with this?

yanfali, 2019-10-17:
Sorry for your trouble, but this really isn't a software issue if you're using stock firmware. I would look for some cold joints, and intermittent connections. It's also possible you have a damaged pro micro. Take a look around the Y, H and N switches and look for suspect soldering.

https://learn.adafruit.com/adafruit-guide-excellent-soldering/common-problems

If this is your first keyboard build, the corne is quite a difficult build and a lot of folks end up repairing it multiple times due to it's inherent fragility. All the best.

nschurmann, 2019-10-17:
Thank you @yanfali  for the quick response! I want to add that this happened even without the switches on. So basically i soldered the diodes, the pro micro and the oled. I flashed the pro micro to "test" that it was working properly... because the flash was successful then i soldered it to the board.

That's a really good resource that you shared, however I don't think it's for my case. This is my second keyboard, the first one is a let's split. So yeah... there is still a lot to learn for me. I will try to reflash it to see if that helps. I bought this keyboard from mechboards.co.uk, De-soldering the pro-micro is very hard, i have tried before and i damaged the pcb of another lets split.

yanfali, 2019-10-17:
so, pro-tip. Socket the pro micro always. They're horrible from a QA perspective. I would take a multimeter out and look for a bad diode too. A bad or reversed diode would cause ghosting. You can get sockets from electronics vendors like digikeys (millmax ulp) and from keebio which can be used to socket a pro micro.

drashna, 2019-10-18:
Does this happen when that half that is having issues, only? 

If so, then it could also be the PCB.  I know that I got a corne PCB from little keyboards that had this sort of issue (to which they shipped the replacement before mine had even arrived).

DavidYangDev, 2019-10-27:
is this issue solved yet? Just wondering because I heard my friend had this issue too( or what he told me)

nschurmann, 2019-10-27:
Hello! I don’t have time to solve this issues as I also create online courses, so an easier path to me was to buy a presoldered pcb. Which I will have to solder the switches only. But apartently was either one of two:
A problem with the pcb
A problem with the pro micro

A safer way, for the future is to socket the pro micros.

closed 2019-10-27 by nschurmann