qmk/qmk_firmware #7074: qmk doctor reports ModemManager is enabled when it isn't

harrywada, 2019-10-18:
## Describe the Bug
When running ```qmk doctor``` on Linux using the repository local CLI, qmk reports ModemManager as enabled when I don't have it installed and there are no systemd unit files associated with it. After looking at ```./lib/python/qmk/cli/doctor.py```, I found the following code:

```python

    elif OS == "Linux":
        cli.log.info("Detected {fg_cyan}Linux.")
        if shutil.which('systemctl'):
            mm_check = subprocess.run(['systemctl', 'list-unit-files'], stdout=subprocess.PIPE, stderr=subprocess.PIPE, timeout=10, universal_newlines=True)
            if mm_check.returncode == 0:
                mm = True
                for line in mm_check.stdout.split('\n'):
                    if 'ModemManager' in line and 'enabled' in line:
                        mm = False

                if mm:
                    cli.log.warn("{bg_yellow}Detected ModemManager. Please disable it if you are using a Pro-Micro.")

            else:
                cli.log.error('{bg_red}Could not run `systemctl list-unit-files`:')
                cli.log.error(mm_check.stderr)

        else:
            cli.log.warn("Can't find systemctl to check for ModemManager.")

```

The current logic for detecting ModemManager obviously doesn't make sense because it will only trigger as not found if every systemd unit file is a unit file enabling ModemManager. This can easily be fixed by swapping the states of ```mm```, so it defaults to ```False``` and only is enabled if a unit file enabling ModemManager is found.

## System Information

 - Operating system: Arch Linux x64
 - QMK Firmware version: master

harrywada, 2019-10-18:
I've opened a pull request: #7075

zvecr, 2019-10-20:
Fixed by #7076

closed 2019-10-20 by zvecr