qmk/qmk_firmware #7094: v60_type_r build fails with RGB enabled

BenRLewis, 2019-10-20:
## Describe the Bug
Unable to build for v60_type_r on latest with RGB support enabled
Attempted with followingghosts and default layout in clean QMK vagrant environment 

## System Information

 - Keyboard:  v60_type_r polestar
   - Revision (if applicable):  
 - Operating system: QMK Dockerfile:latest 
 - AVR GCC version: 4.9.2

 - ARM GCC version: 6.3.1

 - QMK Firmware version: 0.7.44

## Additional Context

Trying to build latest version of QMK for v60_type_r polestar using a clean copy of the vagrant environment in Docker

When I enable RGB, and define `V60_POLESTAR`, the build fails with the following error:
```
Linking: .build/v60_type_r_default.elf                                                              [ERRORS]
 |
 | .build/obj_v60_type_r_default/quantum/backlight/backlight_avr.o: In function `__vector_17':
 | /vagrant/quantum/backlight/backlight_avr.c:259: multiple definition of `__vector_17'
 | .build/obj_v60_type_r_default/keyboards/v60_type_r/v60_type_r.o:/vagrant/keyboards/v60_type_r/v60_type_r.c:155: first defined here
 | collect2: error: ld returned 1 exit status
 |
tmk_core/rules.mk:298: recipe for target '.build/v60_type_r_default.elf' failed
make[1]: *** [.build/v60_type_r_default.elf] Error 1
Make finished with errors
Makefile:555: recipe for target 'v60_type_r:default' failed
make: *** [v60_type_r:default] Error 1
```

fauxpark, 2019-10-21:
Thanks for reporting! I made a quick PR that should hopefully resolve this; see above. It compiles for me now 👍

BenRLewis, 2019-10-21:
Just to confirm, fauxparks commit fixes it, rebuilt the firmware and everything seems good

PR #7099 (2019-10-22): Change V60 Polestar RGB timer to 3

## Description

When Backlight and the custom RGB are both enabled on this board, they both try to use timer 1. The latter should be using timer 3 instead, to avoid this conflict.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #7094

## Checklist

- [x] My code follows the code style of this project.
- [ ] My change requires a change to the documentation.
- [  [... cut]
closed 2019-10-22 by zvecr