qmk/qmk_firmware #7121: ACTION_TAP_DANCE_LAYER_TOGGLE doesn't work?

dky, 2019-10-23:
I'm trying to define leverage the ACTION_TAP_DANCE_LAYER_TOGGLE but it's not working for me. I'm basically using it as so:

[TD_DVORAK] = ACTION_TAP_DANCE_LAYER_TOGGLE(KC_1, 2)

Which should send a number 1 or toggle to layer 2 if double tapped. However during compilation I get:

Compiling: keyboards/kinesis/keymaps/dky/keymap.c                                                  keyboards/kinesis/keymaps/dky/keymap.c:134:16: error: implicit declaration of function 'ACTION_TAP_DANCE_LAYER_TOGGLE'; did you mean 'ACTION_TAP_DANCE_DUAL_ROLE'? [-Werror=implicit-function-declaration]
  [TD_DVORAK] = ACTION_TAP_DANCE_LAYER_TOGGLE(KC_1, layer),
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                ACTION_TAP_DANCE_DUAL_ROLE
keyboards/kinesis/keymaps/dky/keymap.c:134:52: error: 'layer' undeclared here (not in a function)
  [TD_DVORAK] = ACTION_TAP_DANCE_LAYER_TOGGLE(KC_1, layer),
                                                    ^~~~~
cc1: all warnings being treated as errors
 [ERRORS]

Can someone please chime in on what I'm doing wrong?

fauxpark, 2019-10-23:
Looks like the documentation is incorrect and the actual name for that Tap Dance action is `ACTION_TAP_DANCE_TOGGLE_LAYER`, not `LAYER_TOGGLE`. Additionally I think you are not filling in the `layer` argument - it has to be a constant value.

noroadsleft, 2019-10-23:
@fauxpark Looks like you and I both missed this on #6629. :blush: It should have been changed as documented. :man_facepalming:

dky, 2019-10-23:
Hrm, I tried the ACTION_TAP_DANCE_TOGGLE_LAYER function just now:

  [TD_DVORAK] = ACTION_TAP_DANCE_TOGGLE_LAYER(KC_1, DVORAK_LAYER),

created a constant #define DVORAK_LAYER     2 in the header.

avr-gcc (GCC) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  21806	      0	  21806	   552e	.build/kinesis_stapelberg_dky.hex

Compiling: keyboards/kinesis/keymaps/dky/keymap.c                                                  keyboards/kinesis/keymaps/dky/keymap.c:136:16: error: implicit declaration of function 'ACTION_TAP_DANCE_TOGGLE_LAYER'; did you mean 'ACTION_TAP_DANCE_DOUBLE'? [-Werror=implicit-function-declaration]
  [TD_DVORAK] = ACTION_TAP_DANCE_TOGGLE_LAYER(KC_1, DVORAK_LAYER),
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                ACTION_TAP_DANCE_DOUBLE
keyboards/kinesis/keymaps/dky/keymap.c:136:16: error: initialization of 'void (*)(qk_tap_dance_state_t *, void *)' {aka 'void (*)(struct <anonymous> *, void *)'} from 'int' makes pointer from integer without a cast [-Werror=int-conversion]
keyboards/kinesis/keymaps/dky/keymap.c:136:16: note: (near initialization for 'tap_dance_actions[4].fn.on_each_tap')
keyboards/kinesis/keymaps/dky/keymap.c:136:16: error: initializer element is not constant
keyboards/kinesis/keymaps/dky/keymap.c:136:16: note: (near initialization for 'tap_dance_actions[4].fn.on_each_tap')
keyboards/kinesis/keymaps/dky/keymap.c:131:45: error: missing braces around initializer [-Werror=missing-braces]
 qk_tap_dance_action_t tap_dance_actions[] = {
                                             ^
keyboards/kinesis/keymaps/dky/keymap.c:136:16:
  [TD_DVORAK] = ACTION_TAP_DANCE_TOGGLE_LAYER(KC_1, DVORAK_LAYER),
                {{
 };
 }}
cc1: all warnings being treated as errors
 [ERRORS]

dky, 2019-10-23:
Hey guys, disregard my last reply, deleted it, I got it to work by pulling the lastest master. Thank you both for your assistance on this!

closed 2019-10-23 by dky