qmk/qmk_firmware #7130: Adafruit Bluefruit LE SPI Friend Pins

skeltr, 2019-10-23:
I would build a bluetooth keyboard with a AT90USB1286 and for the bluetooth a Adafruit Bluefruit LE SPI Friend (https://www.adafruit.com/product/2633)
But the problem is that i don't know which pins of the module must be connected to which pins of the microcontroller.
Thank you in advance and sorry for my bad english.

noroadsleft, 2019-11-11:
Sorry for the non-response. Did you figure this out?

skeltr, 2019-11-12:
Someone say me that : B1 = SCLK, B2 = MOSI, and B3 = MISO
But can I connect the others pins of the bluefruit to any pins of the at90usb1286 ?
And which pins of the bluefruit are essential and which are not.

fauxpark, 2019-11-12:
You can easily confirm this using the datasheet: http://ww1.microchip.com/downloads/en/devicedoc/doc7593.pdf

On page 3, it says that the SPI pins for the AT90USB64/128 are indeed B1-3.

skeltr, 2019-11-13:
And the others pins (cs, irq, dfu, rst), where i have to connect them ?

fauxpark, 2019-11-13:
According to the docs: https://docs.qmk.fm/#/feature_bluetooth?id=adafruit-ble-spi-friend
B4, E6 and D4 respectively - the DFU pin is for putting the Bluetooth module itself into DFU mode, it's not necessary to connect it to anything unless you're trying to update the firmware, which it can do over the air anyway.

skeltr, 2019-11-14:
Can I change the pins of the cs, irq, rst ?

closed 2020-01-09 by skeltr