qmk/qmk_firmware #7139: Combining keyboard and analog device for perfect analog input?

JonatannossreP, 2019-10-24:
I've this idea that it would be nice to have full X, -X, Y and -Y analog input mapped to your movement keys on a keyboard with regular switches. If you press a key on your keyboard you gain an analog value equal to "full throttle/joystick hold". Then you have a separate analog key or foot padel which you can break with and reduce your speed with. By combining an effectively fixed input with an analog one you could have the benefit of instant direction change as well as precise movement with variable speed.  

Is there some way to program this currently with QMK? Having a fixed (emulated) analog signal on normal keyboard switches, and then have this value multiplied with whatever analog value the analog break would put out?

drashna, 2019-10-26:
Well, mousekeys does something like this.  Not for a joystick, but for the mouse.  

And I think there is a HID Joystick PR actually.

JonatannossreP, 2019-10-26:
It exist a QMK library that allows you to use a Joystick, but I have no idea how to combine analog and a "digital" signal. Neither have I heard of it supporting an analog signal working as a break in any configuration. I don't think you can emulate joystick input with keyboard keys either, and I rather not have an extra program like JoyToKey. Some help would be appreciated. As far as I know, you could potentially do it with JoyToKey if you buy a pedal system, but then I can't make my own and I would've to buy it.

4sStylZ, 2019-10-30:
Did you have checked the PR mentionned by @drashna wich is this one https://github.com/qmk/qmk_firmware/pull/4226 ?

I think this reply to many questions and, if there is exactly a standard implementation for slow / break an analog signal then it's just a question of custom code for your needs.

JonatannossreP, 2019-10-30:
Yeah, I did check it, but as far as I understand breaking and analog input from a non-analog button wasn't a default feature, and since I can't really write custom code it was a dead-end at least for me.

closed 2020-02-27 by stale[bot]