qmk/qmk_firmware #7204: USB enumeration problem of bluepill boards with fake STM32F103 (CS32F103C8T6)

pcsquirrel, 2019-10-30:
## Describe the Bug

Newer bluepill boards from ebay seem to have a counterfeit STM32F103 (aka CS32F103C8T6).
Most times detected while trying to program them, which fails by mismatching chipid.
(STMF103 id: 0x1ba01477 / CS32F103 id: 0x2ba01477)

As per specification the STM device has 64kb Flash, but community has proved that al of them have 128kb in reality. The counterfeit CS32F103 devices have "only" the announced 64kb.

In "tmk_core/common/chibios/eeprom_stm32.h" a flash size of 128kb is assumed (for MCU_STM32F103RB and derivates) which leads to unexpected results when emulated eeprom is read. This happens while USB enumeration which leads to failing USB enumeration.

After changing flash size (FEE_MCU_FLASH_SIZE) to 64 qmk firmware is working flawlessly on CS32F103 cpus.

## System Information

 - Operating system:
   verified on Linux and Windows 10
 - QMK Firmware version:
   0.7.45

yanfali, 2019-10-30:
This one is pretty tough. These chips actually masquerade as a STM parts. Even the stlink software doesn't attempt to deal with these correctly. Not really sure what you want QMK to do about this? If you're using a fake part you sort of have to do the overrides in your keymap directory I would say.

One potential approach would be to add a a specific MCU that specifies CS32F103, instead of pretending these are STM32.

pcsquirrel, 2019-10-31:
After struggling with qmk on this devices i googled a lot and couldn´t find anything helpful. So I dug into the problem and found a solution I would have been glad if some else had figured it out and shared this information. So this bug-report is mainly to be found by others struggling with the same issue.
As the patch is working for me I could provide this patch as a pull request.
I would introduce an additional MCU:
* STM32F103R8 with it´s documented 64kb flash
or
* CS32F103 with it´s real 64kb flash?

noroadsleft, 2019-10-31:
I personally would prefer using the `CS32F103`, especially as I have two of these knockoff BluePills on my desk right now (though interestingly, the chips are branded "CKS").

sigprof, 2020-08-09:
As for the chip name, the `stlink` package uses `CKS32F103` instead of `CS32F103` for those chips — see the [compatible devices list](https://github.com/stlink-org/stlink/blob/v1.6.1/doc/devices_boards.md) and [changelog](https://github.com/stlink-org/stlink/blob/v1.6.1/CHANGELOG.md#v160).

It turned out that I have the `CKS32F103C8T6` chip inside my ST-LINK V2 programmer [from AliExpress](https://www.aliexpress.com/item/32786765274.html) (bought together with a BluePill board, which seems to have a real STM32F103C8T6), but I can't use it for testing without getting another ST-LINK programmer first.

eriqadams, 2021-11-25:
Is there any support for new Bluepill that came with new stm32f103c6t6a chip with 32K Flash size ?, I have tried to burn STM32duino bootloader and flash QMK firmware on to it succesfully but failed to recognize as HID input (Unknown USB Device (Device Descriptor Request Failed)) ?

closed 2022-08-28 by zvecr