qmk/qmk_firmware #7220: Corne Keyboard Issue

pmstyle123, 2019-10-31:
Yesterday I assembled a Corne keyboard. It has two pro micros in it as it's a split keyboard. The left side... when connected by itself works ok. However the right side of the keyboard it's acting weird. I will try to describe it as follows:

When you first plug in the keyboard it works fine... it gets all your keystrokes and you can use it with ease. This keyboard also has some oled panels.
However... randomly, the keyboard will start typing caracters for you. This is very random so i think that, if the assembly was somewhere bridged, then it should be constant... however the keyboard it's usable for a couple of minutes, sometimes only seconds. The keys that are usually randomly pressed are this as in a qwerty keyboard:
Y
H
N

As you can see is the first column of the keyboard... or the 7th column of a complete keyboard... i checked the solder but it seems ok to me.

Have anyone have had a similar issue with this?

drashna, 2019-10-31:
Sounds like there may be a solder bridge somewhere.  Check the back of the PCB, as it can be VERY easy to do this. 

Also, check your diodes and switches, just in case.

pmstyle123, 2019-11-01:
yeah got it!!!

drashna, 2019-11-02:
Did you mean that yo got it working?   Or?

pmstyle123, 2019-11-03:
yeah its working now!!!

closed 2019-11-03 by pmstyle123