qmk/qmk_firmware #7284: The right half of the split keyboard does not work.

dalbodeule, 2019-11-06:
I'm writing new code for my custom keyboard. But there seems to be a problem with this code.

The code is at here. [https://github.com/dalbodeule/qmk_firmware/tree/master/keyboards/rythm_keyboard](https://github.com/dalbodeule/qmk_firmware/tree/master/keyboards/rythm_keyboard)

The problem is as follows. The right module does not work when USB is plugged into the left module. If you plug a USB into the right module, both will not work. At present, there is no 4.4KOhm resistor, so I tried with 1KOhm resistor and 10KOhm resistor.

**However, when I plugged the USB into the right module, the left module did not work.**

If you know the answer to this question, please help.

yanfali, 2019-11-07:
I would look at electrical issues first. Check for shorts on the TRRS connector and that you have a good cable. The most likely explanation is this is not a QMK issue but one with your PCB and/or soldering

dalbodeule, 2019-11-07:
The important thing is that I am in the process of verifying the circuit to order the PCB, and assembled it on the breadboard. There is nothing wrong with the circuit. But the right half doesn't work. Perhaps the Arduino in which this code works doesn't know which module it is, or it seems to be a problem with Arduino itself.

dalbodeule, 2019-11-07:
The problem with I2C communication is solved by using serial communication. But the other problem is that the USB connected module is recognized as a left hand module. I want both modules to be the master.

yanfali, 2019-11-07:
> The problem with I2C communication is solved by using serial communication. But the other problem is that the USB connected module is recognized as a left hand module. I want both modules to be the master.

Ah, there are reports of batches of pro micros with bad vusb detection on the market. There's a software work around that delays startup for 2.5seconds but does resolve the issue 

`#define SPLIT_USB_DETECT`

For a longer explanation as to what is probably going on:
https://medium.com/@keebio/the-case-of-the-wayward-elite-c-73f0fd691f88

yanfali, 2019-11-07:
Feel free to close the issue if this fixes it. All the best

closed 2020-03-06 by stale[bot]