qmk/qmk_firmware #7285: Building with OLED driver fails

dmelliot, 2019-11-06:
When I attempt to build with OLED support, I get various issues.

I'm following the steps at: https://docs.qmk.fm/#/feature_oled_driver?id=oled-driver and I am building using the docker approach.

I first add `OLED_DRIVER_ENABLE = yes` to my `rules.mk` and attempt a build and get the following issue:
```
Compiling: keyboards/converter/usb_usb/custom_matrix.cpp                                           In file included from keyboards/converter/usb_usb/custom_matrix.cpp:39:0:
quantum/quantum.h:146:29: fatal error: oled_driver.h: No such file or directory
 #    include "oled_driver.h"
                             ^
compilation terminated.
 [ERRORS]
 | 
 | 
 | 
tmk_core/rules.mk:377: recipe for target '.build/obj_converter_usb_usb_pro_micro_dme/custom_matrix.o' failed
make[1]: *** [.build/obj_converter_usb_usb_pro_micro_dme/custom_matrix.o] Error 1
Makefile:555: recipe for target 'converter/usb_usb/pro_micro:dme' failed
```

In an attempt to fix this, I updated `quantum.h` and changed this line:
`#    include "oled_driver.h"`
to
`#    include "oled/oled_driver.h"`

This allows the build to progress further, but then fails with:
```
make[1]: *** No rule to make target '.build/obj_converter_usb_usb_pro_micro_dme/oled_driver.o', needed by '.build/converter_usb_usb_pro_micro_dme.elf'.  Stop.
Make finished with errors
Makefile:555: recipe for target 'converter/usb_usb/pro_micro:dme' failed
make: *** [converter/usb_usb/pro_micro:dme] Error 1
```

Any help here would be appreciated.
(Also, if the update to `quantum.h` noted above is required/valid, I can provide that in a PR)

dmelliot, 2019-11-07:
So I've found another keyboard that builds successfully with OLED driver support (`suihankey/rev1`).
This also builds fine if I revert the change to `quantum.h` mentioned above (so clearly this is not needed).

So the question is, does the `converter/usb_usb/pro_micro` build support OLED driver, and if not, how difficult is it to add?

zvecr, 2019-11-07:
I can get it to compile with the following changes:
```diff
diff --git a/drivers/avr/i2c_master.h b/drivers/avr/i2c_master.h
index 0a3b6c508..a4207936d 100644
--- a/drivers/avr/i2c_master.h
+++ b/drivers/avr/i2c_master.h
@@ -20,6 +20,8 @@
 #ifndef I2C_MASTER_H
 #define I2C_MASTER_H
 
+#include <stdint.h>
+
 #define I2C_READ 0x01
 #define I2C_WRITE 0x00
 
diff --git a/keyboards/converter/usb_usb/rules.mk b/keyboards/converter/usb_usb/rules.mk
index b4f37d3ad..1f7cc9667 100644
--- a/keyboards/converter/usb_usb/rules.mk
+++ b/keyboards/converter/usb_usb/rules.mk
@@ -25,6 +25,6 @@ EXTRAKEY_ENABLE  = yes  # Audio control and System control(+450)
 USB_HID_ENABLE   = yes
 CUSTOM_MATRIX    = yes
 
-SRC = custom_matrix.cpp
+SRC += custom_matrix.cpp
 
 DEFAULT_FOLDER = converter/usb_usb/hasu
diff --git a/tmk_core/protocol/usb_hid.mk b/tmk_core/protocol/usb_hid.mk
index e72008097..a6ba51c43 100644
--- a/tmk_core/protocol/usb_hid.mk
+++ b/tmk_core/protocol/usb_hid.mk
@@ -70,4 +70,4 @@ VPATH += $(TMK_DIR)/$(ARDUINO_DIR)/variants/leonardo
 #    Windows doesn't know difference between common/print.h and arduino/Print.h.
 #    On Linux no problem.
 #    Change file name common/print.h to console.h ?
-VPATH := $(TMK_DIR)/common $(VPATH)
+#VPATH := $(TMK_DIR)/common $(VPATH)

```

The main issue is that `USB_HID_ENABLE   = yes` adds extra makefile rules, one of which modifies the code search path.

dmelliot, 2019-11-07:
Thanks for the help here Joel, really appreciated - I've managed a successful build now (although I'm currently away from my test hardware so can't actually test the OLED screen).

In terms of me committing these changes back to master, I can see the first two changes should be ok, but the third change would likely break things for other people - do you see a way to make this change without impacting others?

That said, the build with the OLED driver has me at 98% capacity of the device storage (without any custom layers or OLED logic defined), so it may be a non-starter anyway ....

dmelliot, 2019-11-08:
I think I'll pull the plug on this idea (getting OLED working with the USB_USB converter).
I just can't find enough space to enable it + a couple of layers + have any actual OLED logic without blowing the size limit.

Thanks for the help so far anyway.

closed 2019-11-08 by dmelliot