qmk/qmk_firmware #7307: make: No rule to make target, even when in correct directory?

ArmsAlpha, 2019-11-09:
## Describe the Bug
When running the make command it always comes back saying "No rule to make target '(target)'. Stop. This happens no matter what command I use, even with make commands which are confirmed to work normally. I'm in the correct directory and I believe the commands are valid - I'm practically copy-pasting them. I'm using MSYS2 and I followed the beginner setup guide exactly. I've tried reinstalling everything but it doesn't change anything. https://imgur.com/iqFvUEM

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system: Windows 10
 - AVR GCC version: avr-gcc.exe (AVR_8_bit_GNU_Toolchain_3.6.1_1752) 5.4.0

 - ARM GCC version: arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

 - QMK Firmware version: 0.7.64

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

fauxpark, 2019-11-09:
The keyboard in your image is actually located under the `kingly_keys/` directory, so the make command should be:

```
make kingly_keys/romac_plus:default
```

ArmsAlpha, 2019-11-09:
Wow that worked! Most stuff I found online for fixes specifically said _not_ to do that so it's strange that it works. Either way, thanks a bunch! I was going insane trying to figure it out.

closed 2019-11-09 by ArmsAlpha

fauxpark, 2019-11-09:
Well, our build system is heavily customised specifically for QMK's use case, so a lot of the advice about make commands probably wouldn't apply here.

PR #7308 (2019-11-09): Fix readme make examples for some Kingly Keys boards

## Description

These were apparently missed in the move to the vendor directory.

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #7307

## Checklist

- [x] My code follows the code style of this project.
- [ ] My change requires a change to the documentation.
- [ ] I have updated the documentation accordingly.
- [x] I have read the [**CONTRIB [... cut]