qmk/qmk_firmware #7310: error: listing the stack pointer register ‘sp’ in a clobber list is deprecated

bithead2k, 2019-11-09:
The fix:

diff --git a/os/common/ext/CMSIS/include/cmsis_gcc.h b/os/common/ext/CMSIS/include/cmsis_gcc.h
index d868f2e64..8b72e1492 100644
--- a/os/common/ext/CMSIS/include/cmsis_gcc.h
+++ b/os/common/ext/CMSIS/include/cmsis_gcc.h
@@ -161,7 +161,7 @@ __attribute__( ( always_inline ) ) __STATIC_INLINE uint32_t __get_PSP(void)
  */
 __attribute__( ( always_inline ) ) __STATIC_INLINE void __set_PSP(uint32_t topOfProcStack)
 {
-  __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
+  __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : );
 }

@@ -187,7 +187,7 @@ __attribute__( ( always_inline ) ) __STATIC_INLINE uint32_t __get_MSP(void)
  */
 __attribute__( ( always_inline ) ) __STATIC_INLINE void __set_MSP(uint32_t topOfMainStack)
 {
-  __ASM volatile ("MSR msp, %0\n" : : "r" (topOfMainStack) : "sp");
+  __ASM volatile ("MSR msp, %0\n" : : "r" (topOfMainStack) : );
 }

drashna, 2019-11-09:
```
 - Keyboard:  
   - Revision (if applicable):  
 - Operating system:
 - AVR GCC version: 
<!-- Run `avr-gcc --version` to find this out. -->
 - ARM GCC version: 
<!-- Run `arm-none-eabi-gcc --version` to find this out. -->
 - QMK Firmware version:
<!-- Run `git describe --abbrev=0 --tags` to find this out. -->
```

I'm guessing that this is with gcc 9.x.  If so, then this is a known issue.  #7006 

This is fixed in chibiOS upstream, but .... updating that is non-trivial.   

Additionally, we've seen some other weird behavior with 9.x, and it's not well tested (the release is a month or so old). 

For now, we recommend sticking with 8.3 or older, as these versions are known to be good.

half-cambodian-hacker-man, 2020-01-25:
What about adding this patch downstream in the repo owned by qmk? https://github.com/qmk/ChibiOS

We can always revert if we want to update later on, but it seems like a good temporary solution.

mboedigh, 2020-12-15:
I have this issue with gcc 8.4.0 on ubuntu wsl. I am not even sure I need ChibiOS. Appartently gcc 8.3 is unavailable now (based on dpkg listing).  Is there a different workaround?

closed 2022-08-28 by zvecr