qmk/qmk_firmware #7321: [Bug] IBM XT Converter generally unusable

acid2000, 2019-11-10:
I've been doing a lot of testing between Soarer's Converter, TMK and QMK to find what is best for my IBM 5150. This issue is a way of stopping people wasting their time.

By far the most reliable for daily use is Soarer's converter. It seems to have no issues with any of the three keyboards I tested with. TMK & QMK seem to have persistent issue interpreting the protocol correctly. This eventually results in either phantom key presses or stuck keys.

I tried a number of different power supplies to ensure this wasn't a result of undervoltage. The issues persisted.

TMK has similar issues. I've documented this in https://github.com/tmk/tmk_keyboard/issues/635.
I would say that QMK is much worse.

My recommendation is not to use the XT converter for these keyboards.

## Describe the Bug

Very quickly phantom key presses etc are seem. Keyboard is unusable.

## System Information

 - Keyboard:  
IBM 5150 using the XT converter
 - Operating system:
Ubuntu
 - AVR GCC version: 
avr-gcc (GCC) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
 - ARM GCC version: 
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
 - QMK Firmware version:
0.7.64
 - Any keyboard related software installed? 
No

## Additional Context

acid2000, 2019-11-10:
One thing to add. I didn’t use a pull up resistor. Is QMK correctly setting the pull up resistor states?

I’m using a pro micro.

acid2000, 2019-11-10:
Here's the output of QMK toolbox with me holding down 'q'

HID device connected: QMK XT keyboard converter (FEED:6512:0001)
  > 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 21 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 

A rogue 21 which looks like its an F. After that no key up is sent so my screen i spammed with Fs.

acid2000, 2019-11-10:
So looking at the xt.h source it does look like the internal pullups are being used. Just to make sure I added two 9k ohm pullups on the clock and data lines.

Still the same issues.

acid2000, 2019-11-11:
I've had a look at the XT code in TMK core.

Things I've tried:
- Updating it to correctly read START(0) - No effect
- Updating it to read on the rising edge (I think Soarer's converter does this) - No effect
- Updating it to hold low if the queue is full (the code in the TMK docs did this) - No effect
- Uncommentting the code which checks if the clock is high or low - No effect

acid2000, 2019-11-11:
I now have two 5150's with different manufacture dates. I've also got another 5v ProMicro which i've flashed from Soarers converter to QMK.

Exact same issue. Holding down the Q key for a few seconds results in misreading of the signal coming off the keyboard.

I pretty convinced now it's not either my wiring (tried two ProMicros) or my keyboard (tried two keyboards). I've also tried yet another powered hub to ensure that the keyboard get a rock solid 5v.

There is something wrong with the XT logic in QMK.

acid2000, 2019-11-25:
Replace clock and data with 1k pull ups. Still issues.

barnumbirr, 2020-09-03:
Hey there,

so I just stumbled upon this issue while checking if I could easily add support for my 5150 to VIA.
I'm currently running Soarer's Converter v1.12 on a Chinese Arduino Micro clone and haven't ran into any phantom key presses or stuck keys. I haven't been using any pull up resistors either but keyboard has been running fine as a daily for a few weeks.
@acid2000 let me know if you need me to try out some stuff to help you get to the bottom of this. Being able to run QMK/VIA would be a dream.

Cheers.

acid2000, 2020-09-03:
This bug has been fixed for a while in the most recent TMK core, it needs to be backported to
QMK.

For now run TMK and all is good.

barnumbirr, 2020-09-03:
Ah, my bad. Didn't see this was fixed in TMK 🤭 Guess I'll wait for the backport then I'll try to get VIA to work.

Davines123, 2020-11-09:
It works like a dream on TMK, is there any way to request the backport or is it whoever decides to do it at the time?

DalvikVM, 2021-01-01:
Hello! I think I've made all the necessary changes for the backport, but I need someone to test it for me. Had a sudden move and I carefully packed my xt away.
I'm not sure if the changes will work, but if someone could give it a go and let me know the results, I'd appreciate it!

[Teensy (halfkay) Version](https://drive.google.com/file/d/1Cwy-FvgASKoqFvRem7nN14qW1UJ0KeWJ/view?usp=sharing) 
[Pro Micro (catalina) Version](https://drive.google.com/file/d/1AgjTMkIQPWKrDfdxx6T6W-l0-wkhuCgq/view?usp=sharing) 

@barnumbirr 
@acid2000 
@Davines123

spidey3, 2021-01-01:
Can you open up pull request with your code changes, so we can build and test?

DalvikVM, 2021-01-01:
I just opened a draft pull request: #11384

DalvikVM, 2021-01-01:
I haven't actually done a pull request before, so pardon any errors I might have made in ignorance. The changes I transferred over from TMK were pretty basic and I'm still learning, so hopefully everything will work, but don't get your hopes up too much.... 😬

DalvikVM, 2021-01-10:
For those using TMK with success, are you using the xt_usb or the ibmpc_usb version? 

It looks like hasu made most of his changes to his ibmpc version, which is a good idea in my opinion. Let the old version go. I can try to bring the changes from the ibmpc version over to the xt, but it's looking like it'll be a bit of work. 

If I'm gonna have to work for it, would my efforts be better used in trying to move his new version over to qmk?

The new one combines multiple ibm keyboard converters into one.

Let me know y'alls opinion.

acid2000, 2021-01-10:
ibmpc_usb, the XT code needs to be deleted because it don’t work reliably. Not sure why it is still in the repo.

DalvikVM, 2021-01-10:
Thats what I was thinking. I might be able to make the xt version work, but I dont really think there is a point.

Davines123, 2021-02-22:
Yeah xt_usb on tmk does work but Hasu has moved on to ibmpc_usb at this point. Would there be a way to get someone to port the code over?

DalvikVM, 2021-02-23:
I started working on it, but didnt bother backing it up and ended up losing all of my work recently....
I dont know enough and I dont have the time to get it working, as its not a couple simple tweaks...

Sorry

spidey3, 2021-02-23:
> Yeah xt_usb on tmk does work but Hasu has moved on to ibmpc_usb at this point. Would there be a way to get someone to port the code over?

I was planning on taking a look at it in a few weeks, but I am going to need some detailed pointers on EXACTLY which firmware is working for which keyboard (F-XT DIN connector, F-122 DIN connector, Model M DIN Connector, Model M PS2, etc.).

Davines123, 2021-02-23:
I know the current QMK xt_usb works for an XT protocol Leading Edge DC-2014

spidey3, 2021-02-23:
> I know the current QMK xt_usb works for an XT protocol Leading Edge DC-2014

Hmmm. It's definitely NOT working for my IBM F-XT.

Davines123, 2021-02-23:
Same, my F XT will work for about 3 or 4 keypresses than garbage.

i-am-shodan, 2021-02-23:
QMK xt_usb code is garbage is is unlikely to ever work because it can't keep up with the data rate of the keyboard. TMK has been reimplemented to work around these issues and is the only real option outside Soarer's convertor.

Leading edge/Falling edge should not make a difference. Bin the xt usb code.

spidey3, 2021-02-23:
> TMK has been reimplemented to work around these issues

@i-am-shodan can you point to the changes in TMK that were done to solve the issue, so that they can be ported?

acid2000, 2021-02-23:
It’s the ibmpc_usb converter code

spidey3, 2021-02-23:
> It’s the ibmpc_usb converter code

Does that work with set3 boards like F-122 / M-122?

Davines123, 2021-02-23:
Yes does set 1 - 3

DalvikVM, 2021-02-23:
Yeah, as @Davines123  said, the ibmpc_usb project combines all of the current Ibm converters in TMK.

xt_usb has been more or less abandoned at this point in favor of ibmpc_usb.

I got pretty far by using the QMK xt_usb folder as a basis for the ibmpc_usb project, as you can fairly easily see what changes need to happen for it to start compiling in QMK.

As for the changes that have been made, check tmk's changlog for the entire repo. You can see what has been changed recently.

marfrit, 2021-05-06:
> I started working on it, but didnt bother backing it up and ended up losing all of my work recently....
> I dont know enough and I dont have the time to get it working, as its not a couple simple tweaks...
> 
> Sorry

Got it running, but only made mappings for codeset2/ 102 key ISO as that is what I have... 

https://github.com/marfrit/qmk_firmware

I copied most of the stuff, moved some to matrix.c, had to change initialization and matrix mapping. 
Code and comments are not cleansed yet, I got it to "works for me" state, might work on in some more later.
As I do not own a battleship M, that will need to wait a bit ;)

closed 2022-08-29 by zvecr