qmk/qmk_firmware #7324: [Bug] Can't use ProMicro pin 5 for RGB LEDs

acid2000, 2019-11-10:
I have a WS2812B strip I've attached to PIN 5 of my Pro Micro. I've set RGB_DI_PIN to C6 which I think is the correct pin.
I get a compiler warning:
error: ‘C6’ undeclared (first use in this function)

I've using this as a reference:
qmk_firmware/keyboards/handwired/splittest/promicro/config.h
. 
## Describe the Bug

error: ‘C6’ undeclared (first use in this function)

What #define do I need to make?

fauxpark, 2019-11-12:
I think some more context is needed here. Would you be able to post your code?

acid2000, 2019-11-12:
I’m using master with the default key map. Just compile the XT usb converter.

fauxpark, 2019-11-12:
Ah, I see now. the keyboard's config.h is missing the config_common.h include, but adding that produces a new error, which seems to be about a collision with `E0` in the custom matrix (even though on the 32U4, this pin does not exist; the pindef stuff is not that smart). Prefixing the state enum members with `XT_STATE_` makes it compile, at least. I'll write up a PR in a sec.

Thanks for reporting!

acid2000, 2019-11-12:
I worry that the bigger problem might be that I’m using the XT codebase which is interrupt based. This might be throwing out the timing. I’ll test your PR anyway

PR #7341 (2019-11-13): XT converter: add config_common.h include and fix E0 collision

## Description

The XT converter didn't have `#include "config_common.h"` which is necessary for defining `RGB_DI_PIN`, among other things. Since the custom matrix already defines `E0` as one of the states (even though this is not an actual pin on the 32U4), they had to be prefixed to avoid this collision.

Thanks to @acid2000 for reporting.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Docu [... cut]
closed 2019-11-13 by noroadsleft