qmk/qmk_firmware #7329: [Bug] Planck/Preonic works on boot in linux, but if I replug the keyboard it doesn't send signals

Zawaken, 2019-11-11:
## Describe the Bug
When I boot up my pc (manjaro) with either my Preonic or Planck plugged in the keyboard works totally fine. But if I flash it or try to replug it, it stops working (as in it does not send any signals to the computer). It does make the startup noise and there is movement in dmesg and journalctl output.

This has started happening after I merged in around 190 or so commits into my own fork. Before that I could flash/replug as many times as I wanted

## System Information

 - Keyboard:  Planck/Preonic but might affect more keyboards
   - Revision (if applicable):  Rev6/Rev3
 - Operating system: Manjaro linux
 - AVR GCC version:  9.2.0

 - ARM GCC version:  8.3.0

 - QMK Firmware version: 0.6.435

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
###Journalctl
```
Nov 11 18:10:03 purity kernel: usb 1-6: new full-speed USB device number 18 using xhci_hcd
Nov 11 18:10:03 purity kernel: usb 1-6: New USB device found, idVendor=feed, idProduct=6061, bcdDevice= 0.03
Nov 11 18:10:03 purity kernel: usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 11 18:10:03 purity kernel: usb 1-6: Product: Preonic
Nov 11 18:10:03 purity kernel: usb 1-6: Manufacturer: OLKB
Nov 11 18:10:03 purity kernel: usb 1-6: SerialNumber: 0
Nov 11 18:10:03 purity kernel: input: OLKB Preonic as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:FEED:6061.0029/input/input78
Nov 11 18:10:03 purity kernel: hid-generic 0003:FEED:6061.0029: input,hidraw6: USB HID v1.11 Keyboard [OLKB Preonic] on usb-0000:2a:00.1-6/input0
Nov 11 18:10:03 purity kernel: input: OLKB Preonic Mouse as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.1/0003:FEED:6061.002A/input/input79
Nov 11 18:10:03 purity kernel: input: OLKB Preonic System Control as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.1/0003:FEED:6061.002A/input/input80
Nov 11 18:10:03 purity kernel: input: OLKB Preonic Consumer Control as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.1/0003:FEED:6061.002A/input/input81
Nov 11 18:10:03 purity kernel: input: OLKB Preonic Keyboard as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.1/0003:FEED:6061.002A/input/input82
Nov 11 18:10:03 purity kernel: hid-generic 0003:FEED:6061.002A: input,hidraw7: USB HID v1.11 Mouse [OLKB Preonic] on usb-0000:2a:00.1-6/input1
Nov 11 18:10:03 purity kernel: hid-generic 0003:FEED:6061.002B: hiddev2,hidraw8: USB HID v1.11 Device [OLKB Preonic] on usb-0000:2a:00.1-6/input2
Nov 11 18:10:03 purity systemd-logind[714]: Watching system buttons on /dev/input/event8 (OLKB Preonic System Control)
Nov 11 18:10:03 purity systemd-logind[714]: Watching system buttons on /dev/input/event9 (OLKB Preonic Consumer Control)
Nov 11 18:10:03 purity systemd-logind[714]: Watching system buttons on /dev/input/event10 (OLKB Preonic Keyboard)
Nov 11 18:10:03 purity systemd-logind[714]: Watching system buttons on /dev/input/event6 (OLKB Preonic)
Nov 11 18:10:36 purity kernel: usb 1-6: USB disconnect, device number 18
Nov 11 18:10:37 purity kernel: usb 1-6: new full-speed USB device number 19 using xhci_hcd
Nov 11 18:10:37 purity kernel: usb 1-6: New USB device found, idVendor=0483, idProduct=df11, bcdDevice=22.00
Nov 11 18:10:37 purity kernel: usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 11 18:10:37 purity kernel: usb 1-6: Product: STM32  BOOTLOADER
Nov 11 18:10:37 purity kernel: usb 1-6: Manufacturer: STMicroelectronics
Nov 11 18:10:37 purity kernel: usb 1-6: SerialNumber: 206139722033
Nov 11 18:10:53 purity kernel: usb 1-6: USB disconnect, device number 19
Nov 11 18:10:55 purity kernel: usb 1-6: new full-speed USB device number 20 using xhci_hcd
Nov 11 18:10:55 purity kernel: usb 1-6: New USB device found, idVendor=feed, idProduct=6061, bcdDevice= 0.03
Nov 11 18:10:55 purity kernel: usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 11 18:10:55 purity kernel: usb 1-6: Product: Preonic
Nov 11 18:10:55 purity kernel: usb 1-6: Manufacturer: OLKB
Nov 11 18:10:55 purity kernel: usb 1-6: SerialNumber: 0
Nov 11 18:10:55 purity kernel: input: OLKB Preonic as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:FEED:6061.002C/input/input83
Nov 11 18:10:55 purity kernel: hid-generic 0003:FEED:6061.002C: input,hidraw6: USB HID v1.11 Keyboard [OLKB Preonic] on usb-0000:2a:00.1-6/input0
Nov 11 18:10:55 purity kernel: input: OLKB Preonic Mouse as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.1/0003:FEED:6061.002D/input/input84
Nov 11 18:10:55 purity kernel: input: OLKB Preonic System Control as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.1/0003:FEED:6061.002D/input/input85
Nov 11 18:10:55 purity kernel: input: OLKB Preonic Consumer Control as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.1/0003:FEED:6061.002D/input/input86
Nov 11 18:10:55 purity kernel: input: OLKB Preonic Keyboard as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.1/0003:FEED:6061.002D/input/input87
Nov 11 18:10:55 purity kernel: hid-generic 0003:FEED:6061.002D: input,hidraw7: USB HID v1.11 Mouse [OLKB Preonic] on usb-0000:2a:00.1-6/input1
Nov 11 18:10:55 purity kernel: hid-generic 0003:FEED:6061.002E: hiddev2,hidraw8: USB HID v1.11 Device [OLKB Preonic] on usb-0000:2a:00.1-6/input2
Nov 11 18:10:55 purity systemd-logind[714]: Watching system buttons on /dev/input/event6 (OLKB Preonic)
Nov 11 18:10:55 purity systemd-logind[714]: Watching system buttons on /dev/input/event8 (OLKB Preonic System Control)
Nov 11 18:10:55 purity systemd-logind[714]: Watching system buttons on /dev/input/event10 (OLKB Preonic Keyboard)
Nov 11 18:10:55 purity systemd-logind[714]: Watching system buttons on /dev/input/event9 (OLKB Preonic Consumer Control)
Nov 11 18:11:07 purity kernel: usb 1-6: USB disconnect, device number 20
```
[... 15 lines cut ...]
[  402.260394] hid-generic 0003:FEED:6061.0021: input,hidraw7: USB HID v1.11 Mouse [OLKB Preonic] on usb-0000:2a:00.1-6/input1
[  402.267684] hid-generic 0003:FEED:6061.0022: hiddev2,hidraw8: USB HID v1.11 Device [OLKB Preonic] on usb-0000:2a:00.1-6/input2
[  494.810102] usb 1-6: USB disconnect, device number 14
[  495.569973] usb 1-6: new full-speed USB device number 15 using xhci_hcd
[  495.729282] usb 1-6: New USB device found, idVendor=0483, idProduct=df11, bcdDevice=22.00
[  495.729283] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  495.729284] usb 1-6: Product: STM32  BOOTLOADER
[  495.729285] usb 1-6: Manufacturer: STMicroelectronics
[  495.729286] usb 1-6: SerialNumber: 206139722033
[  509.974845] usb 1-6: USB disconnect, device number 15
[  511.729960] usb 1-6: new full-speed USB device number 16 using xhci_hcd
[  511.889977] usb 1-6: New USB device found, idVendor=feed, idProduct=6061, bcdDevice= 0.03
[  511.889979] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  511.889980] usb 1-6: Product: Preonic
[  511.889980] usb 1-6: Manufacturer: OLKB
[  511.889981] usb 1-6: SerialNumber: 0
[  511.933130] input: OLKB Preonic as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:FEED:6061.0023/input/input68
[  511.986769] hid-generic 0003:FEED:6061.0023: input,hidraw6: USB HID v1.11 Keyboard [OLKB Preonic] on usb-0000:2a:00.1-6/input0
[  511.993218] input: OLKB Preonic Mouse as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.1/0003:FEED:6061.0024/input/input69
[  511.993285] input: OLKB Preonic System Control as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.1/0003:FEED:6061.0024/input/input70
[  512.050052] input: OLKB Preonic Consumer Control as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.1/0003:FEED:6061.0024/input/input71
[  512.050089] input: OLKB Preonic Keyboard as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.1/0003:FEED:6061.0024/input/input72
[  512.050180] hid-generic 0003:FEED:6061.0024: input,hidraw7: USB HID v1.11 Mouse [OLKB Preonic] on usb-0000:2a:00.1-6/input1
[  512.057075] hid-generic 0003:FEED:6061.0025: hiddev2,hidraw8: USB HID v1.11 Device [OLKB Preonic] on usb-0000:2a:00.1-6/input2
```

jackhumbert, 2019-11-11:
Do you know what commit you were using previously where this wasn't an issue?

Zawaken, 2019-11-11:
Hi Jack, 

The commit I was on previously seems to be `c8fd0156186ed08f88ff73a73c00a58004bf217a`

And I did not have any issues then. I can try flashing the default keymap from my branch that still isn't up to date.

Zawaken, 2019-11-11:
Hi again Jack,

I have found the offending commit with git bisect, and it seems to be `cec391df70197a63b12a69c0e08927ecb156e341`

Hope this makes troubleshooting the issue easier.

jackhumbert, 2019-11-11:
If you set `RGBLIGHT_ENABLE = no` in your `rules.mk`, does it change anything?

Zawaken, 2019-11-11:
I have tested this on both the "broken" Preonic and the "broken" Planck, and this does indeed work.

I do think there should be more of a permanent fix instead of this workaround. As new people might run into the same issue.

Thanks, now all my OLKB boards work again.

closed 2019-11-11 by zvecr