qmk/qmk_firmware #7333: Planck EZ suddenly unresponsive/extremely slow

semonje, 2019-11-11:
I'm not quite sure where to post this, so I apologize if this isn't the right place.. :)

**I have a Planck EZ that I've been using for a few months, and today it suddenly stopped reacting to any input (nothing happens).**

After further research, I've discovered that it does in fact react, but I need to hold the key for around 3-4 seconds before that one key is registered. I can't repeat the same key twice, they have to be two different keys in order to work.

I haven't tried to do anything to the firmware for a few months. I upgraded to MacOS Catalina a few weeks ago, without that seeming to affect anything.

I tried to flash it by putting it in bootloader/dfu mode, but as far as I can tell from the QMK toolbox, it doesn't seem to recognise the device as a DFU bootloader. It says `STM32 device connected`. When I try to flash it with a standard .bin for the plank-ez, it seems to work and it plays the little startup jingle. But still, it's unresponsive/extremely slow.

Any help is greatly appreciated! :)

noroadsleft, 2019-11-12:
@drashna ?

drashna, 2019-11-17:
That's very unusual.  Is this an EZ Glow, or just a regular EZ (eg, does it use the rgb matrix). 

Also, are you compiling from the QMK repo, or the ZSA repo? 

Also, do you have autoshift enabled, as it could potentially cause this sort of combination. 

Best yet, do you have your code pushed to github? 

Also:
 - AVR GCC version: (Run `avr-gcc --version` to find this out.)
 - ARM GCC version: (Run `arm-none-eabi-gcc --version` to find this out.)
 - QMK Firmware version: (Run `git describe --abbrev=0 --tags` to find this out.)

Also, does this version have the issue that you're seeing? 
[planck_ez_drashna.bin.zip](https://github.com/qmk/qmk_firmware/files/3855984/planck_ez_drashna.bin.zip)

semonje, 2019-11-17:
Hi! I actually managed to fix it just now. I have a local setup using symbolic links to keep my personal layouts and the latest from QMK up to date, where I seem to have mixed up the `planck/ez` and `planck/rev5` configs. When I made sure to use the latest EZ config, it fixed the issue.

I'm still a bit confused about why it happened without me touching the firmware for 3 months, though..

To answer your questions:
- It's a regular EZ.
- I'm compiling from the QMK repo.
- My code is available here: https://github.com/semonje/planck/tree/master/ez
- Not using autoshift.
- AVR GCC: `v7.4.0`
- ARM GCC: `v7.3.1`
- QMK Firmware: `v0.7.79`

Thanks a lot for answering :)

yanfali, 2019-11-18:
This doesn't really sound like a QMK issue. So I'll be closing this.

closed 2019-11-18 by yanfali

nickfyson, 2022-07-28:
PSA - for anyone who finds their way here when facing this problem after using the ZSA configurator, make sure the standard/glow switch is set appropriately!

[image]

If you pick 'glow' with a non-RGB board things get very weird...